SteveMacenski/spatio_temporal_voxel_layer #43: Voxel grid and costmap stop marking

tasilb, 2018-03-06:
Observed this after the costmap reset was triggered and when there were a lot of voxels left marked. Normal behavior was restored after the grid was purged by another reset or when all previously marked voxels expired

SteveMacenski, 2018-03-06:
probably just fixed

reopen if not true

tasilb, 2018-03-07:
Still not fixed, observed again even with not as many marked voxels

SteveMacenski, 2018-03-07:
this has nothing to do with the number of voxels marked, it has to do with the reset operation apparently.

SteveMacenski, 2018-03-07:
https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/kinetic-devel/src/spatio_temporal_voxel_layer.cpp#L391

https://github.com/ros-planning/navigation/blob/53087ca5242fb40f06c84a6bcc74057c3470eff7/costmap_2d/plugins/obstacle_layer.cpp#L578

SteveMacenski, 2018-03-07:
similar issue appears in the mapping publisher at 1hz. interesting.

SteveMacenski, 2018-03-07:
I think this is from dirty configs, closing again, but may reopen

closed 2018-03-07 by SteveMacenski