SteveMacenski/spatio_temporal_voxel_layer #59: Costmap update from spatio-temporal-voxel-layer

naveedhd, 2018-06-20:
I am using the following config:

```
# costmap_ros
global_frame: /base_link
robot_base_frame: /base_link

update_frequency: 4.0
publish_frequency: 4.0
static_map: false
map_type: costmap
resolution: 0.015

width: 10.0
height: 10.0

rolling_window: true
transform_tolerance: 0.4

plugins:
 - {name: rgbd_obstacle_layer, type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}

rgbd_obstacle_layer:
  enabled:                  true
  voxel_decay:              15    # seconds if linear, e^n if exponential
  decay_model:              0     # 0=linear, 1=exponential, -1=persistent
  voxel_size:               0.05  # meters
  track_unknown_space:      false  # default space is known
  max_obstacle_height:      2.0   # meters
  unknown_threshold:        15    # voxel height
  mark_threshold:           0     # voxel height
  update_footprint_enabled: true
  combination_method:       1     # 1=max, 0=override
  obstacle_range:           3.0   # meters
  origin_z:                 0.0   # meters
  publish_voxel_map:        true # default off
  transform_tolerance:      0.2   # seconds
  mapping_mode:             false # default off, saves map not for navigation
  map_save_duration:        60    # default 60s, how often to autosave
  observation_sources:      rgbd1_mark rgbd1_clear
  rgbd1_mark:
    data_type: PointCloud2
    topic: /detected_obstacles
    marking: true
    clearing: false
    min_obstacle_height: 0.1     # default 0, meters
    max_obstacle_height: 3.0     # default 3, meters
    expected_update_rate: 1.0    # default 0, if not updating at this rate at least, remove from buffer
    observation_persistence: 0.0 # default 0, use all measurements taken during now-value, 0=latest
    inf_is_valid: false          # default false, for laser scans
  rgbd1_clear:
    data_type: PointCloud2
    topic: /detected_obstacles
    marking: false
    clearing: true
    max_z: 7.0                  # default 0, meters
    min_z: 0.1                  # default 10, meters
    vertical_fov_angle: 0.8745  # default 0.7, radians
    horizontal_fov_angle: 1.048 # default 1.04, radians
    decay_acceleration: 5.0     # default 0, 1/s^2. If laser scanner MUST be 0
    voxel_filter: false          # default off, apply voxel filter to sensor, recommend on

```

and getting the voxel layer display as following:
[image]

[image]

My questions are:

1. Is, in the first place, Spatio-Temporal-Voxel-Layer supposed to update costmap_2d? If yes, what could be wrong in the configuration?
2. Voxels appear and disappear which should be part of acceleration and decay but seems to be very non-deterministic and couldn't be set to desirable/understandable behavior.

SteveMacenski, 2018-06-20:
I dont see a map, where is it? Are you running navigation? It looks like you're just visualizing the depth camera pointcloud. You have a set height and width, that seems suspect to me, that's normally defined by the map. 

If you run your config with vanilla voxel grid does it work?

naveedhd, 2018-06-21:
I am using a small test source with just the following:

```
costmap_2d::Costmap2DROS costmap_ros("costmap", tf);
costmap_ros.start();
```
so without navigation..

The costmap is set to `rolling_window` (same of local_costmap in move_base) hence the width-height.

When using the VoxelLayer from costmap_2d, it updates the cost values as following:
[image]

I am supposing that `STVL` should do at the very least update the costmap, if the config and everything else is set up properly.

SteveMacenski, 2018-06-21:
It does indeed update the costmap. I have never tested costmap_2d alone. A PR to fix this behavior is encouraged if this is your use-case.

SteveMacenski, 2018-06-21:
This may be a function of your test rig too, is your code available? I can test on my end as well.

SteveMacenski, 2018-06-21:
I just sat down and made an example using the costmap2dros interface and I have the spatiotemporal voxel grid indeed marking in the costmap, my guess is something is wrong with your configuration. 

[image]

if you look at my recently merged PR, I added this https://github.com/SteveMacenski/spatio_temporal_voxel_layer/tree/kinetic-devel/test folder as my work in validating your issue. You can run that so you have the same thing I tested on. the only other changes I made were removign the static/inflation layers, `static_map` to false and adding a staticly sized map as you did. 

If I were to guess, you either namespaced wrong or forgot the change the topic name.

closed 2018-06-21 by SteveMacenski

naveedhd, 2018-06-25:
I took your test script and ran it (with a stupidly simple dummy point cloud) and indeed it updates the costmap. On playing a bit with config, I found that the problem is when `rolling_window: true` (and `static_map: false`). 

I have added some mocks to reproduce these in [this branch](https://github.com/naveedhd/spatio_temporal_voxel_layer/tree/rolling-test). You can run [this launch](https://github.com/naveedhd/spatio_temporal_voxel_layer/blob/rolling-test/test/minimal_test.launch) with `costmap_type:= global/local` to switch `rolling_window` on and off and test with dummy or real point cloud.

With `rolling_window: false`:
[image]

With `rolling_window: true`:
[image]

SteveMacenski, 2018-06-25:
have you pulled latest? Someone else filed a bug with local costmaps and I patched on Friday.

SteveMacenski, 2018-06-25:
You also didn't fully define your local costmap, namely at minimum `width` and `height` of that local costmap, its likely not showing up due to this. Refer to general guides for configuring the navigation stack, it seems some of your issues are with that.

I'm curious about your use-case. I have submitted STVL to ROSCon this year and I'm gathering datapoints about what folks are using this for.

SteveMacenski, 2018-08-01:
Hi! I'm compiling some photographs of people's robot that have tried or are using the STVL for my upcoming ROSCon 2018 talk on this layer and its use. I'd love if you took a screen shot of your simulations or if you had hardware as an example! thanks!

https://roscon.ros.org/2018/