SteveMacenski/spatio_temporal_voxel_layer #60: Costmap not getting cleared properly

Kapcirt, 2018-06-21:
Hello, I'm using this configuration 

```
rgbd_obstacle_layer:
    enabled:               true
    voxel_decay:           15     #seconds if linear, e^n if exponential
    decay_model:           0      #0=linear, 1=exponential, -1=persistent
    voxel_size:            0.05   #meters
    track_unknown_space:   true   #default space is unknown
    observation_persistence: 0.0  #seconds
    max_obstacle_height:   2.0    #meters
    unknown_threshold:     15     #voxel height
    mark_threshold:        0      #voxel height
    update_footprint_enabled: true
    combination_method:    1      #1=max, 0=override
    obstacle_range:        3.0    #meters
    origin_z:              0.0    #meters
    publish_voxel_map:     true   # default off
    transform_tolerance:   0.2    # seconds
    mapping_mode:          false  # default off, saves map not for navigation
    map_save_duration:     60     #default 60s, how often to autosave
    observation_sources:   rgbd1_mark rgbd1_clear 
    rgbd1_mark:
      data_type: PointCloud2
      topic: /throttle_filtering_points/filtered_points
      marking: true
      clearing: false
      min_obstacle_height: 0.3    #default 0, meters
      max_obstacle_height: 3.0     #defaule 3, meters
      expected_update_rate: 0.0    #default 0, if not updating at this rate at least, remove from buffer
      observation_persistence: 0.0 #default 0, use all measurements taken during now-value, 0=latest 
      inf_is_valid: false          #default false, for laser scans
    rgbd1_clear:
      data_type: PointCloud2
      topic: /throttle_filtering_points/filtered_points
      marking: false
      clearing: true
      max_z: 7.0                  # default 0, meters
      min_z: 0.1                  # default 10, meters
      vertical_fov_angle: 0.8745  # default 0.7, radians
      horizontal_fov_angle: 1.048 # default 1.04, radians
      decay_acceleration: 5.0     # default 0, 1/s^2. If laser scanner MUST be 0
      voxel_filter: true    
```

The obstacles are detected properly and the costmap is updated, but when the points are deleted due to the voxel decay, the costmap portion corresponding to these point is not properly cleared.
Here is a video, it might be more explicit : [https://youtu.be/G7ypkJwLBuM](url)
Did you faced this error ? And if yes, do you where it come from ?

Thanks for the help

SteveMacenski, 2018-06-21:
I'm not sure what you're referencing. I see the white points (I presume those are the voxel grid points not your sensor readings, right?) marking  the costmap in that table in the center. After the acceleration period the points and costmap points disappear.

I think you may be referencing those odd triangular regions after other areas decay that later go away? I can say that I've never seen that before. All of this data looks too perfect, is this a simulation? I notice things updating incredibly slowly, this could be a function of computational load on your computer if you're using gazebo- the depth camera plugin itself really takes a toll. 

Can you confirm this behavior in 'real life'?

SteveMacenski, 2018-06-21:
I think I understand better, those white points are sensor outputs not actually the voxel grid, right? Its very likely there are still points marked in the grid, try visualizing the actual /voxel_grid.

Kapcirt, 2018-06-22:
You are right, this is a simulation. 
For the other comments, the whites points are the voxel grid and not the sensor input and it's really slow because I'm updating the global costmap at a frequency of 5Hz and plublishing it at 2 Hz but now that you said it, it is indeed much slower.

I'll try to test it in real environment to see how it is behaving.

Thanks for your help, I'll keep you updated if I find anything new

SteveMacenski, 2018-06-22:
If this is a simulation issue, certainly we should work to resolve it since many people rely on it, but this was developed primarily for "real world" application

SteveMacenski, 2018-06-27:
I'm going to close this unless there's an update here saying you have issues in real-life. If you're finding that it works in real life fine and you've debuged your simulation environment, we should file a new ticket around simulator testing.

closed 2018-06-27 by SteveMacenski

Kapcirt, 2018-07-09:
I think I found what was wrong, I just needed to have an update frequency lower then the publish frequency. In my previous configuration they were equal and I guess (not sure though) that there was a conflict between the update and publishing of the costmap.

https://youtu.be/5Jl_rIEvC10 

PS : When trying to use the mapping mode, this error popped up 
`what():  Time is out of dual 32-bit range`
From what I read, it is a problem that occurs in simulation because the ros::time type is uint32_t and it cannot express negative value which can happen in simulation ( https://answers.ros.org/question/209452/exception-thrown-while-processing-service-call-time-is-out-of-dual-32-bit-range/ ). Just to inform you in case you wanted to make it simulation friendly.

Thanks for the help

SteveMacenski, 2018-07-09:
Awesome! Glad you got it working. Having the publish higher than the update may result in undefined behaviors- nice catch.

Mapping mode is mostly a toy just to show that it is possible to generate 3D maps of size with low overhead, but I should look into that. The fixes seem minimally simple.

Kapcirt, 2018-07-12:
Hey, I still experiences some issues (the correct behavior was with laser_obstacle_layer and stvl_layer both enabled) and I got one question, what is your static map resolution ?

Kapcirt, 2018-07-12:
Hey, I still got some issues (it is only working when the obstacle_laser_layer and the stvl_layer and enabled together), I was wondering what is your global costmap resolution ? 
Since this parameters is getting set by the static map, I can't really modify it and I was thinking that it might be important

Kapcirt, 2018-07-12:
Here is another video https://www.youtube.com/watch?v=y4ctBYCbiiI&feature=youtu.be from the top view we can see that the clearing is half done at some spot.
Is it possible that it has something to do with the updateBounds that doesn't include points ?

SteveMacenski, 2018-07-13:
The global costmap is the same resolution as the static map, if the static map is included as a layer. 

It looks like you're in the odom frame, not map frame, which may skew what the results you're looking at are. Your resolution doesn't appear to be consistent, but its difficult for me to tell without the voxels visualized at their true size (unless they are and you've set the resolution incorrectly).

SteveMacenski, 2018-08-01:
Hi! I'm compiling some photographs of people's robot that have tried or are using the STVL for my upcoming ROSCon 2018 talk on this layer and its use. I'd love if you took a screen shot of your simulations or if you had hardware as an example! thanks!

https://roscon.ros.org/2018/

Kapcirt, 2018-08-07:
Hi! Sorry for the late answer, here are the picture of the simulation and the hardware on which I tested the STVL.
Good luck for your talk at the ROSCon !

[image]

[image]

SteveMacenski, 2018-08-07:
Awesome! Thanks for the note!