SteveMacenski/spatio_temporal_voxel_layer #78: Cmake error while build from source

tanyouliang95, 2018-08-29:
Having Error while building from source. The problem is mainly on `find_package(OpenVDB REQUIRED) find_package(TBB REQUIRED)`. Although I have already apt-get all those mentioned dependencies, still i encounter cmake  dependencies issue. The error msg is shown below. Thanks for the help!!

``` CMake Error at spatio_temporal_voxel_layer/CMakeLists.txt:33 (find_package):
  By not providing "FindOpenVDB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenVDB", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenVDB" with any
  of the following names:

    OpenVDBConfig.cmake
    openvdb-config.cmake```

SteveMacenski, 2018-08-29:
On what system are your building (OS / hardware)?

SteveMacenski, 2018-08-29:
I tried on 2 freshly provisioned Ubuntu 16.04 machines on Thinkpads without an issue. CI runs with some cloud machines that only use rosdep for dependencies so I might check that. 

```
rosdep install --from-paths src --ignore-src -r -y && catkin build
```
Let me know if your config is different or if you're still seeing issues!

tanyouliang95, 2018-08-31:
Alright, rather than using apt-get for openvdb and tbb, I think rosdep solved it. Thanks!

SteveMacenski, 2018-08-31:
Awesome!

closed 2018-08-31 by SteveMacenski

David9696, 2018-09-24:
Hi,

In my freshly installed ubuntu 16.04 + ros kinetic machine, I tried to build this package from source.

After i keyed in the following command:

rosdep install --from-paths src --ignore-src -r -y && catkin build

I still get the following output.

  Could not find a package configuration file provided by "OpenVDB" with any
  of the following names:

    OpenVDBConfig.cmake
    openvdb-config.cmake

  Add the installation prefix of "OpenVDB" to CMAKE_PREFIX_PATH or set
  "OpenVDB_DIR" to a directory containing one of the above files.  If
  "OpenVDB" provides a separate development package or SDK, be sure it has
  been installed.

Hope you could enlighten me.

Thanks.

discord-intech, 2018-11-06:
I suppose you are using ROS Melodic.

The problem lies in the CMakeList.txt where the CMAKE_MODULE_PATH is set (line 5) :
`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/")`

CMAKE_SOURCE_DIR is set by catkin and does not point to the package's source dir.
Use PROJECT_SOURCE_DIR instead :
`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${PROJECT_SOURCE_DIR}/cmake/")`

SteveMacenski, 2018-11-07:
If you find this to be an error for Melodic, do you mind PRing it to the melodic-devel branch?

discord-intech, 2018-11-08:
Done : #87

SteveMacenski, 2018-11-08:
Merged. Thanks for the contribution!