SteveMacenski/spatio_temporal_voxel_layer #79: "Failed to update mark." and non-working decay in current master, works with same config at commit 2fa4809d

kajep09, 2018-09-17:
Hello and thank you for an awesome package. 
I have been utilizing your package for 3-4 months now and have build it from source (at commit 2fa4809de09f01c0de1b1b9271c1225772d64cc0) on my mobile robots.

This has worked really well with proper acceleration and decay of the voxels, but at the current version (release available via apt-get) with the same config file as before, I can no longer get the decay functionality to work - it seems as if the voxels are added and removed at every iteration no matter the parameters of the layer.
Furthermore, it keeps spamming "Failed to update mark." in the terminal which I never saw with the previous version.

I have used the same data from a rosbag for the comparison so both the data and the configuration is exactly the same and I only get issues with the current version.

I haven't dug too deep into your code or looked at what changed between my working version and the current master - perhaps you could shed a little light on things that might have changed that could lead to the "Failed to update mark." warning?

The config file I have used is the following:

```
rgbd_obstacle_layer:
  enabled:               true
  voxel_decay:           15     #seconds if linear, e^n if exponential
  decay_model:           0      #0=linear, 1=exponential, -1=persistent
  voxel_size:            0.05   #meters
  track_unknown_space:   true   #default space is unknown
  observation_persistence: 0.0  #seconds
  max_obstacle_height:   2.0    #meters
  unknown_threshold:     15     #voxel height
  mark_threshold:        4      #voxel height
  update_footprint_enabled: true
  combination_method:    1      #1=max, 0=override
  obstacle_range:        4.0    #meters
  origin_z:              0.0    #meters
  publish_voxel_map:     true   # default off
  transform_tolerance:   0.2    # seconds
  mapping_mode:          false  # default off, saves map not for navigation
  map_save_duration:     60     #default 60s, how often to autosave
  observation_sources:   picoflexx_clear picoflexx_mark

  picoflexx_mark:
    data_type: PointCloud2
    topic: safecop1_royale_camera/points
    marking: true
    clearing: false
    min_obstacle_height: 0.2     #default 0, meters
    max_obstacle_height: 2.0     #defaule 3, meters
    expected_update_rate: 0.0    #default 0, if not updating at this rate at least, remove from buffer
    observation_persistence: 0.0 #default 0, use all measurements taken during now-value, 0=latest 
    inf_is_valid: false          #default false, for laser scans
    clear_after_reading: false    #default false, clear the buffer after the layer gets readings from it
    voxel_filter: true           #default off, apply voxel filter to sensor, recommend on 
  picoflexx_clear:
    data_type: PointCloud2
    topic: safecop1_royale_camera/points
    marking: false
    clearing: true
    min_z: 0.1                   #default 0, meters
    max_z: 10.0                  #default 10, meters
    vertical_fov_angle: 1.0      #default 0.7, radians
    horizontal_fov_angle: 1.0   #default 1.04, radians
    decay_acceleration: 1.0       #default 0, 1/s^2. If laser scanner MUST be 0
```

Thank you!

kajep09, 2018-09-17:
Hello again

In my quest to create a minimum example to reproduce the issue (in which I reinstalled the package from apt-get) I am in a situation where I can no longer get it to output the warning/error "Failed to update mark.".

However, I did create some comparison videos between my old version (at commit 2fa4809d ...) and the newest release version and there is still something wrong with the voxel decay, or at least it behaves differently between the two different versions.

I have recorded some video from RVIZ with the same rosbag file with all the sensor data and used the configuration from the original issue (above) to compare the output in the voxel grid between the two package versions.

The videos are below with side by side comparisons, with and without voxel_filter: true for the marking observation source.

The video on the left side labeled stvl_old is the old version (2fa4809d) (that I like so much :)) and stvl_new is the newest release version (at time of commenting).

The raw pointcloud2 is colored red and the stvl voxels are colored white (please excuse the noise in the pointcloud that happens sometimes).

What is interesting to me is that on the `stvl_old` videos the voxel_grid is much more dense (more voxels are present) after the sensor has moved away and the voxels are only subjected to the decay clearing. This is especially apparent when the sensor is looking at a solid wall and on the `stvl_new` video there are large gaps in the voxels of the wall (`see video 1 from 1:00`).

Furthermore, in the `stvl_new` video there are some voxels that are never decayed and remain forever, as seen on the return trip at `1:05 in video 1`.

#### With voxel_filter: true on marking
https://viewsync.net/watch?v=JrrGDSpPN2o&t=5&v=0vYElkGS6YA&t=10&mode=solo

#### With voxel_filter: false on marking
https://viewsync.net/watch?v=2b8QN7TCzWA&t=0&v=GErO800CHEw&t=2&mode=solo

Can you shed some light on what (if anything?) changed between the two versions that could elicit this kind of behavior?

SteveMacenski, 2018-09-18:
Hi @kajep09 thanks for filing a ticket! It was very detailed and explicit and we were able to track down the problem quickly. 

You are correct and that's a nice catch that doesn't come up much in our configurations. 

@tasilb has a PR pending (https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/80) that should resolve this issue. I'm curious how you got the Failed To Mark error and what you did to resolve. That's usually not a good thing that you're seeing and would only occur if you were trying to concurrently add things to the data structure (were you editing?). 

P.S. I'm actually giving a talk on this at [ROSCon 2018](https://roscon.ros.org/2018/) in about 1.5 weeks. If you could send me some information about your use-case and an image of your robot, I'd love to high light it in my talk along with others using.

SteveMacenski, 2018-09-19:
PR merged, apt release updates in progress https://github.com/ros/rosdistro/pull/19142

kajep09, 2018-09-20:
Hi @SteveMacenski thanks for the quick fix (I have yet to try it) and the feedback. I have not yet reproduced the "Failed to update mark" error, but I will keep an eye out for it in the future.
I have sent you an email regarding the ROSCon talk - best of luck!

SteveMacenski, 2018-09-21:
Thanks for your input and ticket!

closed 2018-09-21 by SteveMacenski