SteveMacenski/spatio_temporal_voxel_layer #85: Missing documentation on how to launch without navigation stack

Achllle, 2018-10-12:
The readme states that this package is independent of using the ROS navigation stack. I can't figure out looking through the code and readme how to render the voxel output in RViz without having move_base set up. That or I'm missing something ;)

I changed the config files to listen to the right topics (I'm using the RealSense D-series). I'm getting zeros on the topic `/stvl_minimal_test/global_costmap/costmap`.

SteveMacenski, 2018-10-12:
When I reference being able to use the grid data structures and wrappers outside of ROS in general, not just outside of navigation. It's library is dynamically linked and through apt you may link against the datastructure to create new applications. I use this library for other projects including an integrated slam solution. On top of this grid is the layer, which is what one might use in a navigation problem in the navigation ecosystem. 

I don't support (currently) use of the layer, as opposed to the grid, outside of the navigation stack. The issue you would run into with that is not having to do with STVL, but costmap_2d itself. If you know costmap_2d is pretty easy to run this all independently but its out of the scope of this particular project to teach folks the basics of the costmap_2d project. A reference implementation can be seen in the test directory or in reading move_base.  

Given your naming, I'm assuming you're using my test files in the test directory. Those do work out of the box, I have used this to test minor changes and address issues brought up by other users as a common test-ground. I'd make sure everything was hooking up correctly (topics going where they should be, see costmap publishing, then visualizing).

Achllle, 2018-10-13:
Yes, I am using your test files. All topics are connected as they should be as far as the readme and config state but it doesn't work. I also tried building from source but am running into the same issue as #78

SteveMacenski, 2018-10-14:
You'll have to provide more information in how it doesn't work or what you expect.

As to the build issue, I've never run into it and I've built it across probably 50 computers at this point. It's interesting to hear this issue but I'll also need more information (how did you install openvdb/tbb, what hardware / OS are you running, etc)

SteveMacenski, 2018-11-08:
@Achllle did #87 fix your issue?

Achllle, 2018-11-10:
I do not know. I decided to move to a different package

SteveMacenski, 2018-11-10:
Closing issue

closed 2018-11-10 by SteveMacenski