SteveMacenski/spatio_temporal_voxel_layer #94: Costmap layer is not cleared when using spatio_temporal_voxel_layer

zoidzilla, 2018-11-20:
Hello,

I have the following inflated spatio_temporal_voxel_layer configuration, which uses a 3D camera onboard a mobile robot:

operators:
   - name: inflate_depth
    type: inflate
    params:
      layer: depth_layer
      output_layer: depth_inflated
      inscribed_radius: 0.1
      inflation_exponential_rate: 5.0
      inflation_radius: 0.3

depth_layer:
  global_frame: map
  robot_base_frame: base_link
  transform_tolerance: 0.4

  update_frequency: 5.0
  rolling_window: true

  resolution: 0.025
  origin_x: 0.0
  origin_y: 0.0
  width: 7.0
  height: 7.0

plugins:
    - {name: depth, type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}
    #- {name: depth, type: "costmap_2d::ObjectDetectionLayer"}

  depth:
    enabled:               true
    voxel_decay:           20     # seconds if linear, e^n if exponential
    decay_model:           0      # 0=linear, 1=exponential, -1=persistent
    voxel_size:            0.05   # meters
    track_unknown_space:   true   # default space is unknown
    mark_threshold:        0      # voxel height
    update_footprint_enabled: true
    combination_method:    1      # 1=max, 0=override
    publish_voxel_map:     false   # default off
    transform_tolerance:   0.2    # seconds
    observation_sources:   depth_clear depth_mark
    depth_mark:
      data_type: PointCloud2
      topic: camera/depth/filtered
      marking: true
      clearing: false
      min_obstacle_height: 0.2     # default 0, meters
      max_obstacle_height: 1.2     # defaule 3, meters
      expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
      min_z: 0.5                   # default 0, meters
      max_z: 3.0                   # default 10, meters
      voxel_filter: true           # default off, apply voxel filter to sensor, recommend on
      vertical_fov_angle: 0.7854   # default 0.7, radians
      horizontal_fov_angle: 1.0472 # default 1.04, radians
      decay_acceleration: 5.      # default 0, 1/s^2. If laser scanner MUST be 0
      obstacle_range: 5.0
      observation_persistence: 0.0 # default 0, use all measurements taken during now-value, 0=latest
      clear_after_reading: false   # default false, clear the buffer after the layer gets readings from it
    depth_clear:
      data_type: PointCloud2
      topic: camera/depth/filtered
      marking: false
      clearing: true
      min_obstacle_height: 0.2     # default 0, meters
      max_obstacle_height: 1.2     # defaule 3, meters
      expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
      min_z: 0.5                   # default 0, meters
      max_z: 3.0                   # default 10, meters
      voxel_filter: true           # default off, apply voxel filter to sensor, recommend on
      vertical_fov_angle: 0.7854   # default 0.7, radians
      horizontal_fov_angle: 1.0472 # default 1.04, radians
      decay_acceleration: 5.       # default 0, 1/s^2. If laser scanner MUST be 0

Unfortunately, as the following video demonstrates, when I turn the robot in another direction, the part of the map that is not currently updated by the point cloud (colored pointcloud has been movied to another location), is decaying only partially but not disappearing completely. Are you aware of such a problem?

https://youtu.be/R3fYxBVcOMY

Thank you!

SteveMacenski, 2018-11-20:
@zoidzilla please edit your post to use markdown, its challenging to read your configurations

SteveMacenski, 2018-11-20:
What are you visualizing, exactly? It looks like that point cloud is not from the voxel layer but your sensor itself. Please show the underlying voxel grid.

SteveMacenski, 2018-11-21:
Please publish the voxel map and display that - that's a more useful thing - the raw sensor data isn't as interesting :)

zoidzilla, 2018-11-21:
Hi again,

This is another video with the voxel grid displayed. It seams that although the voxel grid behaves correctly, the costmap2d underneath it is not updated accordingly.  The 3D camera we are currently using is the IFM O3D303 and the voxel layer receives the pointcloud at around 17Hz.  

https://youtu.be/svkE9Jej_TQ

Below there is the full configuration that we have right now, which combines a lidar as well. However, what I display to you here (and the resulting videos), is the output of the cosmap related only to the voxel grid layer (depth_layer).

Please let me know what you think.

 ```
plugins:
   - name: static_layer
     type: canvas::plugins::StaticLayer
 
   - name: lidar_layer
     type: canvas::plugins::ObstacleLayer
 
   - name: depth_layer
     type: canvas::plugins::WrapperCostmap2D
 
 operators:
   - name: inflate_static
     type: inflate
     params:
       layer: static_layer
       output_layer: static_inflated
       inscribed_radius: 0.1
       inflation_exponential_rate: 5.0
       inflation_radius: 0.3
 
   - name: inflate_lidar
     type: inflate
     params:
       layer: lidar_layer
       output_layer: lidar_inflated
       inscribed_radius: 0.1
       inflation_exponential_rate: 5.0
       inflation_radius: 0.3
 
   - name: inflate_depth
     type: inflate
     params:
       layer: depth_layer
       output_layer: depth_inflated
       inscribed_radius: 0.1
       inflation_exponential_rate: 5.0
       inflation_radius: 0.3
 
   - name: merge_lidar_inflated_and_depth_inflated
     type: merge
     params:
       layer_a: lidar_inflated
       layer_b: depth_inflated
       output_layer: sensors_inflated
       combination: max
 
   - name: merge_static_inflated_and_sensor_inflated
     type: merge
     params:
       layer_a: static_inflated
       layer_b: sensors_inflated
       output_layer: combined_inflated
       combination: max
 
   - name: publishers
     type: publish
     params:
       layers:
         - static_layer
         - lidar_layer
         - depth_layer
         - static_inflated
         - lidar_inflated
         - depth_inflated
         - sensors_inflated
         - combined_inflated
 
 static_layer:
   map_topic: /map
   first_map_only: false
 
 lidar_layer:
   resolution: 0.025
   size_x: 10.0
   size_y: 10.0
   laser_scan_topic: /laser/scan
 
 depth_layer:
   global_frame: map
   robot_base_frame: base_link
   transform_tolerance: 0.4
 
   update_frequency: 5.0
   rolling_window: true
 
   resolution: 0.025
   origin_x: 0.0
   origin_y: 0.0
   width: 7.0
   height: 7.0
 
   plugins:
     - {name: depth, type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}
  depth:
     enabled:               true
     voxel_decay:           20     # seconds if linear, e^n if exponential
     decay_model:           0      # 0=linear, 1=exponential, -1=persistent
     voxel_size:            0.05   # meters
     track_unknown_space:   true   # default space is unknown
     mark_threshold:        0      # voxel height
     update_footprint_enabled: true
     combination_method:    1      # 1=max, 0=override
     publish_voxel_map:     true   # default off
     transform_tolerance:   0.2    # seconds
     observation_sources:   depth_clear depth_mark
     depth_mark:
        data_type: PointCloud2
        topic: camera/depth/filtered
        marking: true
        clearing: false
        min_obstacle_height: 0.2     # default 0, meters
        max_obstacle_height: 1.2     # defaule 3, meters
        expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
        min_z: 0.5                   # default 0, meters
        max_z: 3.0                   # default 10, meters
        voxel_filter: true           # default off, apply voxel filter to sensor, recommend on
        vertical_fov_angle: 0.7854   # default 0.7, radians
        horizontal_fov_angle: 1.0472 # default 1.04, radians
        decay_acceleration: 5.      # default 0, 1/s^2. If laser scanner MUST be 0
        obstacle_range: 5.0
        observation_persistence: 0.0 # default 0, use all measurements taken during now-value, 0=latest
        clear_after_reading: false   # default false, clear the buffer after the layer gets readings from it
     depth_clear:
        data_type: PointCloud2
        topic: camera/depth/filtered
        marking: false
        clearing: true
        min_obstacle_height: 0.2     # default 0, meters
        max_obstacle_height: 1.2     # defaule 3, meters
        expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
        min_z: 0.5                   # default 0, meters
        max_z: 3.0                   # default 10, meters
        voxel_filter: true           # default off, apply voxel filter to sensor, recommend on
        vertical_fov_angle: 0.7854   # default 0.7, radians
        horizontal_fov_angle: 1.0472 # default 1.04, radians
        decay_acceleration: 5.       # default 0, 1/s^2. If laser scanner MUST be 0

 footprint: [[-0.08, -0.275], [-0.08, 0.275], [0.72, 0.275], [0.72, -0.275]]
 footprint_padding: 0.01

 global_costmap_layer: combined_inflated
 local_costmap_layer: sensors_inflated

```

SteveMacenski, 2018-11-21:
Ah we're getting somewhere. Can you try again with only the STVL layer running (no video necessary, just "does it work or not")? 
So actually remove and turn other things off, even if if the visualizations are just of this. It could be a bad interaction with the layers causing it.

zoidzilla, 2018-11-21:
Hi,

With everything removed from config and only the depth_layer active, it is exactly the same behavior as before. :/ 

Please advice further.

Thank you!

SteveMacenski, 2018-11-27:
Can you send a bag file with tf, tf static, and your sensor readings?

Also can you try clearing costmaps after you do this and tell me if the grid disappears under the area the grid remains marked after the points are gone?

SteveMacenski, 2019-02-25:
@zoidzilla any bag file? 
Another user emailed me saying that the clear costmap *recovery behavior* did not work, which I understand now how it doesnt. Is this what you were experiencing as well or are you talking about costmap clearing as a service / reset?

opendvd, 2019-02-28:
> Hello,
> 
> Thank you for your answer. As you can see in the layer configuration I posted previously, the created depth_layer uses a plugin of type "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer", which is the actual spatio temporal voxel grid implementation. Below you can see another video of the same scenario but with the inflation removed this time. So what you visualize is the actual raw pointcloud (from 3D sensor) inserted into the sptio_temporal_voxel_layer and its costmap2d projection. So when the mobile platform changes its orientation, the previously occupied costmap2d grid points that are no longer updated by any raw pointcloud points, are only partially deleted after the decay time (voxel_decay), which in our case is set to 20sec.
> 
> https://youtu.be/YnCBZ9HbL0s
> 
> Please let me know if this is indeed an issue or there is something else going on.
> 
> Thank you in advance.

Hello, what i what to know is how the STVL displays, is it uses rviz? 
The local_costmap_params.yaml  only has the rgbd_obstacle_layer, the costmap is regular, but the STVL can't display, so i want to know how to display it just like your video on youtube.
Thank you very much.

SteveMacenski, 2019-02-28:
set `publish_voxel_map` to true and look for the voxel grid topic in rviz. You can display the local costmap with the Map type

opendvd, 2019-03-02:
> set `publish_voxel_map` to true and look for the voxel grid topic in rviz. You can display the local costmap with the Map type

Thanks for your reply! It can display now.

SteveMacenski, 2019-03-04:
Closing issue from no response and no other users reporting

closed 2019-03-04 by SteveMacenski