SteveMacenski/spatio_temporal_voxel_layer #96: Dynamically disable the layer

MarcelSoler, 2018-11-22:


SteveMacenski, 2018-11-27:
This could be implemented a number of ways

MarcelSoler, 2018-11-29:
Hi! 

- I could stop sending data to the layer but I don't want to change the camera driver. I want to take advantage of the fact that the camera driver does not consume resources if you don't have any active subscribers in the camera topics.

-  I think that with the dynamic reconfigure changes you have suggested will work really well. 

Thank you!

SteveMacenski, 2018-11-29:
Ok, veimox hasnt responded in awhile - if you wanted to implement some of this, you could branch off his and finish it and submit a new PR

MarcelSoler, 2018-12-03:
I'll wait until next week , if he hasn't responded I'll do it!

MarcelSoler, 2018-12-05:
Hello I am starting to take a look at Veimox branch and I have a suggestion:
- I think having the enabled param for each observation sources can be a good way to control them.
This way you can just  enable / disable the observation buffers with the dynamic reconfigure or even a service.

Let me know what do you think of it.

SteveMacenski, 2018-12-05:
Sure that sounds like a reasonable idea, but I'm not sure how that would be done clearly and generally given that you don't necessarily know how many layers there are, or what order they're in. If you have a clean solution to that, I'd love to hear it and if it seems reasonable, a very good contribution

Also looking at your comment on the other PR, C++17 is supported not sure why you don't think it is. It's not default in kinetic, but it works fine. I've been running this on many robots for about a year now without an issue.