SteveMacenski/spatio_temporal_voxel_layer #99: compillation errors with catkin_make

MarcelSoler, 2018-12-10:
On ubuntu16.06 kinetic with cakin_make:

```
CMake Error at spatio_temporal_voxel_layer/CMakeLists.txt:41 (find_package):
  By not providing "FindOpenVDB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenVDB", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenVDB" with any
  of the following names:

    OpenVDBConfig.cmake
    openvdb-config.cmake

  Add the installation prefix of "OpenVDB" to CMAKE_PREFIX_PATH or set
  "OpenVDB_DIR" to a directory containing one of the above files.  If
  "OpenVDB" provides a separate development package or SDK, be sure it has
  been installed.
```
And with catkin build it works.
Thank you,

SteveMacenski, 2018-12-10:
Did you rosdep install your dependencies like in the instructions? 
This compiles in CI and our machines, please make sure you have all the dependencies required through rosdep

SteveMacenski, 2018-12-11:
Here's the rosdep install instructions. I should frankly remove any other install instructions, I find that people installing them independently has issues. https://github.com/SteveMacenski/spatio_temporal_voxel_layer#install-from-source

MarcelSoler, 2018-12-12:
I have all dependencies installed with rosdep using` catkin build` compiles. But if i try to use `catkin_make` it doesn't find OpenVDB.

nickvaras, 2018-12-12:
Just to chip in, I saw a similar thing when I tried to put stvl as a third party subfolder within a larger workspace. In the end I had to give it its own workspace and, after making, sourcing stvl separately. Not sure if related, but maybe.

SteveMacenski, 2018-12-12:
Can you try changing the line 

`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_SOURCE_DIR}/cmake/") `

to 

`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${PROJECT_SOURCE_DIR}/cmake/")`

to see if that resolves @nickvaras @MarcelSoler

MarcelSoler, 2018-12-13:
Solved! I just commit it in pr #98

commit adb7347b (2019-01-03): Dynamic reconfigure proposal continuing (#98)

* Added primitive dynamic reconfigure configuration

* Added naive callback for dynamic reconfigure

* Using positive values in voxel_decay due to dynamic reconfigure
limitations

* using only positive values in voxel decay

* Using raw poiters to follow project's convenction

* We use c++17 so we have templated lambdas!

* Added publishing option to dynamic reconfigure

* adresing comments

* added enabled param for each observation buffer (wip)

- added a service to enable or disable each observation buffer
- c++14 istead of 17 kinetic compatibl [... cut]
closed 2018-12-13 by MarcelSoler