SteveMacenski/spatio_temporal_voxel_layer #100: Error when using it with melodic

Patrick-Lascombe, 2018-12-11:
Hello Steve !
I'm using your layer with a robot that runs with ROS melodic, I used the correct branch and it compiled fine but when I tried to run it. At some point move base crashed because of this error :
`/opt/ros/melodic/lib/move_base/move_base: symbol lookup error: /home/keylo/Ros_ws/devel/lib//libspatio_temporal_voxel_layer.so: undefined symbol: _ZN3tf212getTimestampIN3pcl10PointCloudINS1_8PointXYZEEEEERKN3ros4TimeERKT_ ` and I didn't managed to crack it.
Have you encountered this error at some point ? And if yes, would you know a fix ?

Thanks for the help

SteveMacenski, 2018-12-11:
Hi Patrick,

Unfortunately we don't run Melodic at the moment, a few folks implemented and tested the melodic devel branch that I trusted should work correctly. 

It looks like the melodic branch is missing some dependencies [here](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/CMakeLists.txt#L63), can you try adding `tf2_geometry_msgs tf2_ros tf2_sensor_msgs` to that list and the package.xml file, clean the workspace, run rosdep again, and rebuild? It's clearly a linking issue if we're actually compiling fine. That usually eliminates the need to debug code and more about the CMakeLists. I'd put a 40% chance that resolves it outright. If so, please let me know or PR those changes. 

If that doesn't work, I'll need to spin something up to debug further. Let me know how it goes

Patrick-Lascombe, 2018-12-12:
I tried what you suggested by the error is still there. It may be due to tf or pluginlib getting changes during the migration to melodic : http://wiki.ros.org/melodic/Migration. I'll investigate a little bit more on that today

SteveMacenski, 2018-12-12:
It reads to me like a problem in PCL `getTimestampIN...pcl...PointCloudIN...PointXYZ...ros..Time` in the observation buffer, but not sure - its definitely a linking error though. Maybe the version of PCL that ships with Melodic has some different needs?

Patrick-Lascombe, 2018-12-20:
By using c++filt to demangled the error message and it seem that `tf2::getTimestamp<pcl::PointCloud<pcl::PointXYZ> > ` is undefined and causing the issue. But I still don't know which package I should add as a run dependency to correct it.

SteveMacenski, 2018-12-20:
It appears to be tf2_geometry_msgs, which was one of the packages missing in the package.xml and the CMakeLists as a catkin_depend. I would try to add it there as well and see if that resolves. It's not in the package xml or under`CATKIN_DEPENDS` in the CMakeLists file

https://docs.ros.org/api/tf2_geometry_msgs/html/c++/namespacetf2.html

Patrick-Lascombe, 2018-12-21:
Nevermind the point cloud was the problem. I forgot that my PC was XYZRGB, so when the measurement buffer tried to transform the PC from the camera to the global frame the type didn't match. I just tuned the camera so it is just XYZ and it is not crashing anymore.
Thanks for the help !

Patrick-Lascombe, 2018-12-21:
I did that and I also added tf2_conversions since point_cloud_ptr is defined in the hpp but same result..

SteveMacenski, 2018-12-21:
Another good place to look is at PCL: do you have it installed and is it the correct version that matches with melodic

Patrick-Lascombe, 2018-12-26:
Yep, I got all the pcl deb installed with the 1.8 version, the one available on Ubuntu Bionic

SteveMacenski, 2018-12-28:
Ok I did some digging, Kinetic (which is the branch I personally support and deploy) runs PCL 1.7 and it looks like Melodic bumps up to PCL 1.8 (which is super annoying that that didn't appear in the migration guide!) 

There's a link below which talks about changing the CMakeLists to `find_package(PCL 1.8 ..)` to specify a version but since we want 1.7 NOT 1.8 you have installed I suspect, you want to change the `find_package(PCL 1.7 REQUIRED)` and install PCL 1.7 alongside PCL 1.8. 

https://answers.ros.org/question/289264/how-do-you-upgrade-pcl-to-the-current-release/

Patrick-Lascombe, 2019-01-03:
I have tried what you suggested but there is still the same error..
Here is the folder included in PCL_INCLUDE_DIRS : `/usr/local/include/pcl-1.7/usr/include/eigen3/usr/include/usr/include/ni/usr/include/openni2/usr/local/include/vtk-8.0` The STVL is using the correct version pcl-1.7. 

Is it possible that other changes didn't appeared in the migration guide ?

SteveMacenski, 2019-01-03:
Can you send me your modified CMakeLists file?

Patrick-Lascombe, 2019-01-04:
Sure here is the CMakeLists file
[CMakeLists.txt](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/files/2726308/CMakeLists.txt)

SteveMacenski, 2019-01-04:
Do you get any warnings?

It compiles for me `find_package(PCL 1.743.1 EXACT)` when I use an obviously wrong version with a warning like

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'PCL' but neither 'PCL_INCLUDE_DIRS' nor
  'PCL_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:62 (catkin_package)

```

You need the `find_package(PCL 1.7.2 EXACT REQUIRED)` to enforce the use of that version only. I checked on my side 1.7.2 looks right, what apt spits out is 1.7.2-14build1

Patrick-Lascombe, 2019-01-07:
I had this warning when I used find_pacakge(PCL 1.7 EXACT), but it corrected when I switched to find_package(PCL 1.7.2 EXACT REQUIRED).
Do I need to also download in parallel the pcl-ros library compatible with 1.7.2 ? I have not done this part so it might be why the mistake was not corrected.

SteveMacenski, 2019-01-07:
Ok, the warning is bad, you need the REQUIRED tag to use it correctly. By "corrected" I'm not sure what you mean, you mean its working now? Or just compiling without the warning?

If we need an older version of PCL ROS, I would recommend instead finding the function causing the issue (which above it olooks like you found) and find a different way around it for the Melodic branch. 2 versions of PCL is annoying but doable, 2 versions of that and pcl_ros is a little too much for just 1 function issue between them. 

You can see the migration of another buffer from Kinetic -> Melodic below. I'd recommend just changing over to the methods they use as a "we know it works" migration to make this all moot and just use the 1.8.X with melodic

https://github.com/ros-planning/navigation/blob/kinetic-devel/costmap_2d/src/observation_buffer.cpp
https://github.com/ros-planning/navigation/blob/melodic-devel/costmap_2d/src/observation_buffer.cpp

Patrick-Lascombe, 2019-01-15:
Ok, I have made some progress. It is working with my base version of PCL (PCL 1.8).
From what I understood there is 2 problems : 
- the pcl_ros::transformPointCloud function used in STVL is not compatible with tf2
- the solution to transform PC with tf2 explained here https://answers.ros.org/question/262179/transform-pointcloud-with-tf2/ and in the costmap_2d links are not compatible with pcl_ros::PointCloud datatype.

The fix I found for now is to create a tf::Transform from the tf2_buffer values and use it in the ros_pcl::transformPointCloud function.

There is still some error but it has more to see with our robot config than with the STVL.
I'll make a PR when I have a stable version.

Thanks a lot for your help !

Patrick-Lascombe, 2019-01-18:
I've created the PR. I'm not really used to creating PR's so if you have any input on how I can improve it I'm open to it.

SteveMacenski, 2019-01-18:
Awesome! I Just reviewed it, there's a few changed I'd like you to make to ensure we're not adding dependencies we don't need and streamlining some of the conversions

SteveMacenski, 2019-02-12:
merged

closed 2019-02-12 by SteveMacenski