SteveMacenski/spatio_temporal_voxel_layer #102: map save duration subtraction

salsicha, 2018-12-20:
When I set the "map_save_duration" to any number greater than "0" (zero), I get this error when launching from a launch file:

[ INFO] [1545335823.065952288]: global_costmap/rgbd_obstacle_layer's global frame is map.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range

From what I understand, two time values are being subtracted and resulting in a negative number which can't be represented in an unsigned container.

SteveMacenski, 2018-12-20:
Please attach your configuration file. I just tried on my robots locally and it works fine with the default parameters (60s). What OS, software version, and ros distribution are you using? This might happen if you specified a _negative_ value. There's no subtraction of this value- there's only a ros time now() - last time which is always positive unless you're going backwards in time. The duration is set as a `ros::Duration(value)` on initialization, so it would crash immediately if you had a non-positive value.

SteveMacenski, 2018-12-21:
haha ok, glad you figured it out :)

closed 2018-12-21 by SteveMacenski