SteveMacenski/spatio_temporal_voxel_layer #110: Measurment buffer has lag in melodic release

Patrick-Lascombe, 2019-02-14:
When testing more thoroughly my PR, I noticed that the voxels were having some lag [video here](https://youtu.be/bvFFtAZBnmw).

It comes from this line (156) in the measurement buffer 
` geometry_msgs::TransformStamped tf_stamped = _buffer.lookupTransform(_global_frame, cloud.header.frame_id.c_str(), ros::Time::now(), ros::Duration(0.1));`
It takes ~10-15 ms to complete, which create a clot in the buffer.

I'll investigate this further and see how it can be solved

SteveMacenski, 2019-02-14:
@Patrick-Lascombe try as 

```
  geometry_msgs::TransformStamped tf_stamped = _buffer.lookupTransform(_global_frame, cloud.header.frame_id.c_str(), ros::Time(0));
```

SteveMacenski, 2019-02-22:
@Patrick-Lascombe any update?

doisyg, 2019-02-25:
The time at which the frame coordinate transform was requested was wrong. This PR fixes it: https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/115
I replaced it by the timestamp of the measurement.

However there is still a serious performance issue caused by this line: 
https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/36916b39fa8c273094e6adfe2d5b736194096de1/src/measurement_buffer.cpp#L96
It takes 0.03s to perform the pcl::fromPCLPointCloud2 transformation, against 0.002s on the same machine under Kinetic. I am not sure of anything but this seems due to PCL (which actually was updated from v 1.7 to 1.8 between Kinetic and Melodic)

SteveMacenski, 2019-02-25:
Looks good to me, merged!

closed 2019-02-25 by SteveMacenski