SteveMacenski/spatio_temporal_voxel_layer #112: 3d Lidar frustum fidelity

nickvaras, 2019-02-20:
The binary release of the stvl has the CONEPADDING value (line 41 of three_dimensional_lidar_frustum.cpp) set to zero (the default value), but testing suggests that it doesn't accurately represent the real frustum. The original assumption is that all observations originate from a single point, but I'm guessing that, in the case of a vlp16 for example, there's an array of lasers that has a non-zero height. Therefore, I wanted to propose bringing this value out so it's configurable on the yaml file.

Since this parameter doesn't exist for the depth camera frustum, I'd thought I'd ask first if this sounds reasonable before proceeding.

nickovaras, 2019-02-20:
My branch "vfov_padding" should work. PR should follow soon.

SteveMacenski, 2019-02-20:
This sounds very reasonable, PRs welcome

nickovaras, 2019-02-20:
#113

closed 2019-02-20 by nickovaras