SteveMacenski/spatio_temporal_voxel_layer #124: Melodic-devel error on lambda function

Patrick-Lascombe, 2019-05-29:
Following the #123, I had the following errors that appears in the DynamicReconfigureCallback
```
/home/patrick/stvl_ws/src/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_layer.cpp:586:45: error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14
   auto updateFlagIfChanged = [&update_grid](auto& own, const auto& reference)
                                             

/home/patrick/stvl_ws/src/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_layer.cpp:586:62: error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14
   auto updateFlagIfChanged = [&update_grid](auto& own, const auto& reference)
```
My g++ version is 7.3.0 so it should have c++14/c++17.
I do not have a lot of experience concerning lamda functions, could it be a requirement or something that my computer is missing ?

Thanks for your help

SteveMacenski, 2019-05-29:
Mhm, that's odd - it compiles fine for me. Is this on a clean rebuild? If you remove the C++17 compiler flag in the CMakeLists does it work for you then? 

Frankly, I don't much like lambdas so it may be worth just breaking out into a function if its going to cause issues.

MarcelSoler, 2019-05-30:
Hi! I am currently migrating to Melodic, when I can I'll test fresh install. And I'll share any things I encounter.

MarcelSoler, 2019-05-31:
Hi It also happened to me in the robot, it fails when linking. I also tried c++14 and c++17.

I am working on it! I'll pull request if I find a solution and we can discuss the best one. Right now I got rid of the landas fast quick and dirty and works. When I have something better I'll  start a pull request.

SteveMacenski, 2019-05-31:
A templated function with a `nonconst bool& state` would work as well (e.i. `T getState<T>(const T& input, bool& state)`) to get rid of the lambda. More than happy with that solution. I'm a little confused why you both see that and I and the author of that PR didn't. Can you give me an insight on the hardware you're running it on?

MarcelSoler, 2019-05-31:
@SteveMacenski  I have tested it with two different computers both clean installation ubuntu 18, one msi and one industrial one. In kinetic it worked is a bit strange...

Anyway I think  the pull request is a better solution because logically speaking is cleaner. Let me know what do you think.

SteveMacenski, 2019-05-31:
May be a CMake version thing then

PR #125 (2019-06-02): Fix melodic compilation
Motivation:
Fix #124 , getting rid of lamdas.

Explanation:
Lamdas are not really necessary in this function, dynamic reconfigure has a way to know what group of parameter has been modified, a bitmask called level.
The inspiration idea credit comes from my colleagues from  [beta_robots](http://www.beta-robots.com/) .

Changes made:
- Separated reconfigure parameters in 2 levels [0,1].
- If any parameter level 1 is modified, in the
callback the received level value will be 1 and we'll update grid.
closed 2019-06-02 by SteveMacenski