SteveMacenski/spatio_temporal_voxel_layer #129: 3D Lidar - Voxels are created but the costmap layer is not updated correctly.

kajep09, 2019-06-17:
Hi @SteveMacenski 

I am using the current kinetic release of STVL and trying it out with a simulated 3D Lidar in Gazebo. The problem I am facing is that I get a lot of populated voxels from the point cloud of the lidar, but the costmap is not being updated correctly (or at all?).

My config is basically the same as your VLP16 example:
```
rgbd_obstacle_layer:
  enabled:                  true
  voxel_decay:              15    # seconds if linear, e^n if exponential
  decay_model:              0     # 0=linear, 1=exponential, -1=persistent
  voxel_size:               0.05  # meters
  track_unknown_space:      true  # default space is known
  max_obstacle_height:      2.0   # meters
  unknown_threshold:        15    # voxel height
  mark_threshold:           0     # voxel height
  update_footprint_enabled: true
  combination_method:       1     # 1=max, 0=override
  obstacle_range:           5.0   # meters
  origin_z:                 0.0   # meters
  publish_voxel_map:        true # default off
  transform_tolerance:      0.2   # seconds
  mapping_mode:             false # default off, saves map not for navigation
  map_save_duration:        60    # default 60s, how often to autosave
  observation_sources:      rgbd1_mark rgbd1_clear
  rgbd1_mark:
    data_type: PointCloud2
    topic: /os1_cloud_node/points 
    marking: true
    clearing: false
    min_obstacle_height: 0.3    # default 0, meters
    max_obstacle_height: 3.0     # default 3, meters
    expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
    observation_persistence: 0.0 # default 0, use all measurements taken during now-value, 0=latest
    inf_is_valid: false          # default false, for laser scans
    voxel_filter: false          # default off, apply voxel filter to sensor, recommend on
    clear_after_reading: true    # default false, clear the buffer after the layer gets readings from it
  rgbd1_clear:
    data_type: PointCloud2
    topic: /os1_cloud_node/points 
    marking: false
    clearing: true
    max_z: 8.0                  # default 0, meters
    min_z: 0.0                  # default 10, meters
    vertical_fov_angle: 0.523   # default 0.7, radians. For 3D lidars it's the symmetric FOV about the planar axis.
    vertical_fov_padding: 0.05    # 3D Lidar only. Default 0, in meters 
    horizontal_fov_angle: 6.29  # 3D lidar scanners like the VLP16 have 360 deg horizontal FOV.
    decay_acceleration: 5.0     # default 0, 1/s^2. 
    model_type: 1               # default 0, model type for frustum. 0=depth camera, 1=3d lidar like VLP16 or similar
```
If I start my navigation with STVL and inflation layer enabled as the only thing, I get the following (Red boxes are STVL voxels, white squares is 2D laser scan and rainbow dots are 3D laser scan)

[image]
 :

If I then enable the standard obstacle layer for the 2D laser scanner, I get the following (only costmap updates around the white squares as expected):

[image]

I tried to create a small video to show this and also moved the robot around a bit so it can be seen that the voxels are actually being updated and decayed correctly.

One weird thing that I encountered (which can be seen in the video around 0:58) is that if I enable the obstacle layer for 2D, and then disable it again, there is some residual information on the costmap, but I am unsure if this is from STVL or some other bug?
[RVIZ video](https://youtu.be/Bfnzm3bJMok)

Overall, I am unsure where to look for things that could have gone wrong because the voxel-part seems to be working, but not the costmap part. I hope you might have some ideas.

P.S. Is dynamic_reconfigure supposed to be working in the latest kinetic release?

nickvaras, 2019-06-17:
Can you share your costmap layer definitions? Typically kept on a separate yaml file, e.g.  costmap_local.yaml or something of the sort.

SteveMacenski, 2019-06-17:
Have you also made sure the topics your trying to look at are correct? If you have multiple configs, maybe you’ve namespaced them differently. 

Does the costmap show up when you disable voxel map publishing? 

If there is voxels appearing and disappearing, the sensor data is getting into the layer and the costmap is calling on the layer to update, so this seems situational

nickvaras, 2019-06-17:
From the video it seems that you do not apply inflation to the stvl layer, so if shown in rviz, it would not appear with the corresponding "bubbles" that you are probably expecting

kajep09, 2019-06-17:
Thanks for the quick response both of you. 
I'll try to give  you some more information about my config and the things I've tried.

First of all, my TF tree is the following (mainly map -> odom_comb -> base_footprint) and my  costmap setup is the following:

[image]

```
global_costmap:
  robot_base_frame: base_footprint
  global_frame: map
  static_map: true
  update_frequency: 3.0
  publish_frequency: 3.0
  inflation:
    cost_scaling_factor:  3.0  
    inflation_radius:     0.6  
```
```
local_costmap:
  robot_base_frame: base_footprint
  # global_frame: odom_comb
  global_frame: map
  static_map: false
  rolling_window: true
  resolution: 0.05
  inflation:
    cost_scaling_factor: 10.0  
    inflation_radius:     0.6 
  width: 4.0
  height: 4.0
  origin_x: 0.0
  origin_y: 0.0

```

If I use the same STVL settings as in the original post, and add the plugin to the global costmap, everything works fine (costmap colored by global costmap below all the red voxels). 

[image]

However, I cannot get it to work properly with the local costmap. My current local costmap settings are below:
Using either `odom_comb` or `map` as the global frame works fine with either the standard obstacle layer or your own [nonpersistent_voxel_layer](https://github.com/SteveMacenski/nonpersistent_voxel_layer) - the local costmap is updated correctly.

However, I cannot get any config to work with STVL and local costmap updates. I think there is something going on with improper transformations, but it's really hard to explain and reproduce in a consistent way. I think a video (https://www.youtube.com/watch?v=fUGDsP60Zu4&feature=youtu.be) shows it better than I can explain it, but with a really big local costmap, I can actually see the local costmap getting updated in the correct shape when comparing to the red voxels of STVL. However, the costmap info from STVL is not following the robot_base_frame correctly.

The video shows that STVL works correctly for the global costmap and at 0:32 I toggle the local costmap on, but the  layer is updated at the incorrect place. To somewhat sanity check the local_costmap params, I toggle (at 00:45) the nonpersistent voxel layer on to see that it updates (based on the white laser scanner dots) at the correct places. Then I drive around a bit and it looks like there is some wrong, weird, non-linear, transformation that makes the layer update at the incorrect place?

Any ideas? Appreciate the help :)

nickvaras, 2019-06-18:
Is the tf from base_link to os1_sensor correct?

Will check my setup and reply in the morning.

nickvaras, 2019-06-18:
In my setup I do not specify 
```
  origin_x: 0.0
  origin_y: 0.0
```
Try loosing them and let me know if it helps,  maybe they are offseting the rolling window...

nickvaras, 2019-06-18:
The rolling window in your video seems to be 10x10m, even though you set it to be 4x4... are your parameters loading correctly?

nickvaras, 2019-06-18:
The rolling window does not seem to be rolling, it keeps showing the 10x10m area from the bottom left corner where the map frame is... Does it work if you set ` global_frame: odom_com` instead?

nickvaras, 2019-06-21:
Any updates on this @kajep09 ? Please let me know if you were able to make it work.

kajep09, 2019-06-21:
Hi guys,

Sorry I haven't had time to respond for a few days. I've done some more testing that I'll try to summarize:
* My `local_costmap` setup works fine with the rolling window and `global_frame` either `map` or `odom_comb` for both the standard navigation stack obstacle layer or Steve's nonpersistent voxel layer
* STVL works fine with my `global_costmap` setup and both the voxels and the costmap update correctly
* I cannot get any `local_costmap` setup to work with STVL - like @nickovaras says, it looks like the rolling window doesn't work correctly here or there is some other frame / transformation issue, but I can't see why. The voxels from STVL update correctly, but the costmap is not updated accordingly.

* To make sure it has nothing to do with the TF tree and the simulated 3D lidar, I have converted the 2D LaserScan (that works correct with obstacle / nonpersistentvoxellayer) to PointCloud2 and use that to evaluate STVL, however I still have the same issues.

I have recorded a rosbag of all the relevant data and tried to create a minimal launch file with my setup. I hope you can run that, otherwise maybe you could test it with a move_base config of your own and remap to my topics if possible? I am very curious how I messed my config up or if there is something else at play.

https://wetransfer.com/downloads/22b3fb8ba1558a411a3374ab3bf44e4920190621184011/efa31791aef135c769635f0b09731bcf20190621184011/3cb882https://wetransfer.com/downloads/22b3fb8ba1558a411a3374ab3bf44e4920190621184011/efa31791aef135c769635f0b09731bcf20190621184011/3cb882

The following should work if you extract everything to the same folder.
```
rosparam set /use_sim_time true
rosrun map_server map_server maze.yaml 
roslaunch move_base.launch
rosbag play stvl_pc2_debug.bag --clock
rviz -d navigation.rviz
```

Most of the topics in the rosbag are self explanatory I think - but `cloud_filtered` is the same as `b_scan`, just as a PointCloud2. So you can use those 2 topics to compare local_costmap changes if you want. 

I will try to play around with a few more things and see if I can make sense of something and record a video. 

Like I said in the OP I am using the current kinetic release version - 3D / 360deg lidar is supposed to work in this version right? Just want to make sure that we are not overlooking something stupid.

nickvaras, 2019-06-21:
Hi @kajep09 , I succesfully use the Kinetic STVL in both simulation and actual robots, so I know for a fact that it works, at least with the right configuration. But I cannot rule out the possibility of a bug or something working in an unintended way, so thanks for the thorough feedback, I'm looking into it to find out why this is happening and hopefully I'll have an answer sooner rather than later.

kajep09, 2019-06-21:
Sorry I somehow pasted the link twice! I've fixed it in the previous post.

I recorded this video using the rosbag I linked earlier. As you can see I toggle the nonpersistentvoxellayer on and off, and the costmap is updating correctly for that. But there is a weird duplicate of the costmap information translated to another place in the rolling window and this only shows up when I have STVL enabled. After the robot is done rotating in place I drive it around a bit (around 0:40) and I can't really explain what happens with the costmap.

https://www.youtube.com/watch?v=u-6j2NIf8Bk&feature=youtu.be

nickvaras, 2019-06-21:
Did you try this?:

>  Does it work if you set `global_frame: odom_comb` instead

(for both the global and local costmaps)

nickvaras, 2019-06-21:
7 seconds into your video, it almost seems that just bringing the rqt_reconfigure window up makes the artifact appear.. makes no sense...

kajep09, 2019-06-21:
Actually the video was recorded with global_frame: odom_comb, but I can't find any consistent difference between the two options. 
If you have the option to replay the bag file yourself and play with rqt_reconfigure then try to play with width and height. These parameters seem to switch where the artifacts show up, but I can't really make sense of it.

I am not at all emotionally attached to my costmap config, so if you have an example of a working configuration with STVL I would love to see it :)

SteveMacenski, 2019-06-25:
Hi guys,
Just scanning over this thread. I’m currently on vacation with extremely limited internet access (think top of a mountain in Hungary). I will not be able to address anything there in a meaningful way until mid-July. My feeling though from everything above its use configuration since I havent heard this from other users

SteveMacenski, 2019-08-24:
Any updates here?

closed 2019-09-25 by SteveMacenski