SteveMacenski/spatio_temporal_voxel_layer #137: Plans for ROS 2?

tasilb, 2019-08-02:
What needs to get done to release STVL as  ROS 2 package?

SteveMacenski, 2019-08-02:
Are you moving to ROS2?

Right now the state of perception in Navigation2 (ROS2's navigation client, very different from the existing navigation stack) is up in the air, I was going to wait until that was hammered out before migrating.

If you have a more immediate need for a Dashing build of STVL, I can help guide you through the migration process but I don't have the cycles this month (probably) to take too much of an active role in it

SteveMacenski, 2019-08-18:
It doesn't sound like from my discussion on navigation2 that the costmap API is going to change, so porting this would be relatively simple.

SteveMacenski, 2019-12-12:
As an update, I'm in the middle of a port. Its 80% done but the last 20% will take some time. I can push the branch to collaborate if someone's interested in taking on a couple features from my todo list

SteveMacenski, 2019-12-13:
ROS2 branch is up.

It generally needs testing but otherwise its the first stab at a port

SteveMacenski, 2019-12-14:
Complete and tested in simulation

closed 2019-12-14 by SteveMacenski