SteveMacenski/spatio_temporal_voxel_layer #138: Detect Dropoff/cliff/edges?

mastash3ff, 2019-08-18:
Is this package able to be configured to detect Dropoff/cliff/edges?  If so, how would a sample configuration look like?

SteveMacenski, 2019-08-18:
You can configure the bottom of registered voxels to be below the ground plane to detect dropoffs, but then you'd need to modify the code to detect the actual ground plane to segment out (so they don't appear as obstacles). 

You're probably better off detecting those items from the sensor streams themselves or implementing another costmap layer. However if you've like to implement it on top of this framework, I would accept that as a PR.

SteveMacenski, 2019-08-18:
I'm closing this ticket since it's not a bug or a feature request specifically in line with this project, but feel free to keep the dialog going.

closed 2019-08-18 by SteveMacenski

madgrizzle, 2020-09-23:
This project has solved so many problems for me, thank you for putting this together. 

>it's not a bug or a feature request specifically in line with this project

I'm new to the ROS world (only started learning it 9-months ago) and though I agree its not a bug, I don't understand why cliff detection seems to be ignored in STVL and pretty much everything else for that matter (not a criticism, just an observation).  The focus always seems to be on detecting obstacles above ground and not equally or more dangerous obstacles such as 'missing' ground.  Last night, I discovered my map was missing part of the stairway railing when I had walked in front of the robot and it calculated an alternate path that took it into the stairwell and back out through the gap in the railing (not realizing the stairwell was a stairwell).  Fortunately I was able to stop it when I saw it turn in the direction of the stairs, but it could have been disastrous (obviously, I'll fix the map to stop that from happening)

I recognize it's not a simple problem to solve because there aren't points in the pointcloud if it can't see the step and you have to create some algorithm to detect it.  Is this why it's not specifically in line with the project?  I just feel this seems to be ignored by most modules when it's actually pretty important.  Again, not a criticism, this is a great module.

madgrizzle, 2020-09-23:
I looked at the code and it seems that it filters pointclouds using PCL filters.  I'm wondering if this is where any cliff detection should/could be done (in a new or modified PCL filter).