SteveMacenski/spatio_temporal_voxel_layer #141: clearing at the edge of the frustum

nickvaras, 2019-10-21:
I find myself needing some sort of way to clear the voxels just on the edge of the frustum. Due to the discretization inherent to a voxelized grid, I've noticed that even a frustum that has been perfectly calibrated to the sensors real physical frustum, is prone to get some readings that fall very near the edge. So far the way I plan to address this is to wiggle the robot in place so as to move the frustum a but and clear those pesky erroneously marked voxels, but a neater approach would be to be able to modify the "padding" of the frustum on the fly. Alternatively, some sort of raytracing along the edge approach would also help... Anyway, before I start any work on this, I thought it might be worth sharing this and asking whether anyone has seen or experienced something of the sort on their applications.

nickvaras, 2019-10-21:
Looking at issue #5, I  wonder if this would achieve what I propose. Having a secondary frustum for the same sensor, inactive most of the time, of similar but slightly inflated dimensions to the working frustum, that I could call to achieve the clearing a smidge past the edge of the frustum, where noisy voxels tend to accumulate and loiter and throw a party every now and then...

SteveMacenski, 2019-10-21:
You address a few things here:

- Needing a way to clear along edge of frustum. This could be done here, technically, but if sensor specific, why not add edge removal filter or similar to your sensor filtering chain?
- Secondly, you could in practice actually just set the FOV of your sensor larger than the actual FOV by a few degrees (specs sat 1.047, make it 1.07 or something) to clear out the edges better. Remember spec sheets for FOV usually have an error tolerance. That would be "padding" to the frustum. Not currently on the fly reconfigurable, but welcome a PR to do so
- `raycasting`, right there I'll stop you and say that isn't really in the scope of this project. There is some work I've done not yet publicly on STVL2.0 that does incorporate raycasting + decay but that's not for this version. I'm also 95% sure that raycasting for the issue you describe would make it a whole lot worse, that's a bad direction especially for edge of frustum points

SteveMacenski, 2019-10-21:
Your other suggestion would work, but I'd challenge you on maybe just increasing the frustum a _little bit_ for general use might do the trick. I think what you're running into is the frustum variation, which no calibration method is going to mask.

However if you wanted to try 2 frustums and activate a clearing frustum that's slightly larger on command, I think that functionality is already exposed for you

nickvaras, 2019-10-21:
I was about to try the second approach actually, I'll share results when I have them. I tried a larger-than-physical FOV a long time ago, but I want to avoid under-clearing voxels as much as I want to avoid over-clearing, as over-clearing can lead to small obstacles sneaking their way into the blind spot unmarked, and potentially getting run over by an angry robot.

nickvaras, 2019-10-21:
What do you mean with frustum variation? Not familiar with that concept...

SteveMacenski, 2019-10-21:
Most high quality cameras will list their FOV as 75 degrees +/- 3 degrees or something. Whats listed isnt necessarily the truth and in configuring packages with FOV values I tend to add some padding like you suggest but for the generally used configuration

nickvaras, 2019-10-21:
Oh, ok. You meant just the spec tolerance. So the issue I'm having the padding approach is that potentially valuable information can be lost that way. That's why I'm exploring the possibility of putting up with more voxels marked and then clearing selectively instead. Which brings me to 

>  I think that functionality is already exposed for you

I can't find the parameter that would let me turn frustums on/of on the kinetic version. I know I can enable/disable the entire layer, but not sure about individual frustums. Any cues?

nickvaras, 2019-10-21:
This is what I'm getting from dynparam:
`{'publish_voxel_map': False, 'map_save_duration': 60.0, 'voxel_decay': 90.0, 'mapping_mode': False, 'update_footprint_enabled': True, 'enabled': True, 'voxel_size': 0.05, 'combination_method': 1, 'track_unknown_space': True, 'decay_model': -1, 'groups': {'publish_voxel_map': False, 'parent': 0, 'voxel_decay': 90.0, 'mark_threshold': 0.0, 'combination_method': 1, 'track_unknown_space': True, 'groups': {}, 'id': 0, 'map_save_duration': 60.0, 'name': 'Default', 'parameters': {}, 'update_footprint_enabled': True, 'enabled': True, 'voxel_size': 0.05, 'mapping_mode': False, 'state': True, 'decay_model': -1, 'type': ''}, 'mark_threshold': 0.0}`

SteveMacenski, 2019-10-21:
https://github.com/SteveMacenski/spatio_temporal_voxel_layer/tree/kinetic-devel#enabingdisabling-observation_sources-real-time

Another user added this service

nickvaras, 2019-10-21:
:heart::heart::heart::heart::heart:

nickvaras, 2019-10-21:
I guess the next step would be to set an initial state for each frustum, other than disabling the edge-clearing frustum at startup

SteveMacenski, 2019-10-21:
Yeah, if you wanted to make a param `enabled` for each that can be set on bringup, I'll merge that. So that you dont have to start in a non-ideal state

nickvaras, 2019-10-21:
PR'd

SteveMacenski, 2019-10-21:
gets get a melodic PR too and i'll squash merge

SteveMacenski, 2019-10-21:
We good to close this?

closed 2019-10-22 by nickovaras