SteveMacenski/spatio_temporal_voxel_layer #147: Are the toggle_enabled frustum services working?

nickvaras, 2019-10-22:
Are the toggle_enabled frustum services working?
Calling the service to enable/disable frustums seems to having no effect at all. Just wanted to survey other users out there just to make sure I'm not going crazy...

SteveMacenski, 2019-10-22:
The steps I'd debug with:

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/spatio_temporal_voxel_layer.cpp#L375
Add a print statement to make sure your sevice calls are being made through

It also just came to my attention in writing this that the user only actually implemented it for Pointcloud2 types not laser types, which might be part of your problem:
https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/spatio_temporal_voxel_layer.cpp#L255

They also pushed back the `_observation_subscribers` twice in PointCloud2 which will mess with the logic...

nickvaras, 2019-10-22:
So I 'm using pc2 types, but had no effect visible from calling the service. But I'll have a deeper look and sort this out.

SteveMacenski, 2019-10-22:
Ugh, yeah sorry, this was a user submitted patch, I didn't do this otherwise I could probably just suss out the problem from reading the code.

Let me know if you  get stumped or need another pair of eyes

nickvaras, 2019-10-22:
Thanks! Like you said,   `      _observation_subscribers.push_back(sub);` is repeated on lines 260 and 281. Still digging...

SteveMacenski, 2019-10-22:
Though if you patch it, I'd appreciate if you extended to the laser types & fixed that dual subscriber push back.

nickvaras, 2019-10-22:
Duh, so I see my issue here. The service is working perfectly fine to enable or disable **observation** sources, meaning the marking observations, but no effect on the clearing. At least I'm understanding the issue better. Also, `IsEnabled()` seems to be used correctly.

SteveMacenski, 2019-10-22:
`IsEnabled` is used correctly, just really never usefully. 

I consider clearing an observation source, sure, we throw out the data, but we wouldn't want to clear without data coming in!

SteveMacenski, 2019-11-27:
@nickvaras can this be closed?

nickvaras, 2019-11-27:
I'm still working on it. Had to put it aside due to more urgent things, but
still intend to get clearing frustums to be toggable.

SteveMacenski, 2021-03-10:
Any update on this? I know this was a _long while_ back

nickvaras, 2021-03-11:
I'll close this for now, as I haven't yet implemented this the way I wanted it (which had as ultimate objective giving me a way to clear the noisy edge of the frustum by switching between frustums of slightly different dimensions, and/or having additional decay "zones" beyond just inside/outside frustum). I would still like to implement this some day, but I has fallen waaay down my to-do list.  Thanks for checking!

closed 2021-03-11 by nickvaras