SteveMacenski/spatio_temporal_voxel_layer #156: Costmap refresh rate issue

blablebli-robots, 2020-01-09:
Hello,

I am trying to run your stvl code on an ARM based development board. I am currently getting around 0.3Hz with a depth camera running at 10Hz. Any pointers to where to start looking will be greatly appreciated, I wish to use this stack for navigation purposes (avoidance).

Also, What are the minimum system requirements to run this software for real time avoidance?

Thank you!

SteveMacenski, 2020-01-09:
I’ve heard of folks running this at 5hz on a embedded ARM processor, are you sure the issue is STVL and you’re using the binary install or optimized build? 

Does using the voxel layer run in your requirements? I have a paper coming out soon that shows that STVL is an order of magnitude faster for many sensors being processed compared to a raycasting voxel method.

blablebli-robots, 2020-01-10:
Thank you for your reply,

STVL was built from source, but dependencies were installed from binary packages. A quick look from the profiler shows roscpp/pthreads as possible culprits. Will keep digging.

The navigation stack (slam+LiDAR+IMU+odom) runs fine without Pointcloud2 messages, as soon as I try to bring in Pointcloud2 into standard VoxelLayer or STVL then I run into refresh issues. With the VoxelLayer I cannot even see the layer being updated.

Cheers.

SteveMacenski, 2020-01-10:
Are you on a release build? Catkin will do debug builds by default. That could have a substantial impact.

blablebli-robots, 2020-01-10:
Yes it is a Release build, thanks for pointing that out. I have a quad core ARM, and camera set to VGA resolution, so I still don't understand what in my setup is causing such drop in performance given the info you have of some ARM running at 5hz.

Thank you again for all your help. I am currently rebuilding dependencies from source to see if there is any improvement. Will keep you posted.

blablebli-robots, 2020-01-13:
Well, after rebuilding dependencies and retesting I'm still running into the same issues, no improvement, my current setup is not cutting it. Thank you for all your support.

SteveMacenski, 2020-01-13:
@blablebli-robots so are there any action items here? Are you going to increase compute or not use pointclouds in STVL or voxel layer? 

Sounds like you just need more compute. Closing but will reopen if I'm misinterpreting. 

(also happy to hear that mine at least runs with voxel layer doesn't even process ;-) )

closed 2020-01-13 by SteveMacenski