SteveMacenski/spatio_temporal_voxel_layer #159: unable to locate minimal_test node

tekrajchhetri, 2020-03-02:
Hi,

I am having some issue with spatio_temporal_voxel_layer while trying an example minimal_test.launch

`roslaunch spatio_temporal_voxel_layer minimal_test.launch 
`
It is unable to locate the minimal_test node and it seems this attribute is mandatory.

ERROR: cannot launch node of type [spatio_temporal_voxel_layer/minimal_test]: can't locate node [minimal_test] in package [spatio_temporal_voxel_layer]

I installed this spatio_temporal_voxel_layer in ubuntu 18 and ros melodic 
`sudo apt-get install ros-kinetic-spatio-melodic-voxel-layer`

I also have a question,  what is the purpose of this minimal_test node and what are other types of nodes available? And where can I find more information about it?

 Thank you.

SteveMacenski, 2020-03-02:
See cmakelist: https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/CMakeLists.txt#L126

This is a minimal test for me to use to test PRs and have a common interface for helping debug peoples setup issues. I do not think this has any value to you or other users. I do not include it in the debians intentionally to not increase the binaries for production systems.

tekrajchhetri, 2020-03-02:
Hi,
Form the installation I did understand it was for test only as it was not included but when I try running launch file (shown below), it was giving me the error I highlighted above.

<launch>
    <node name="test_run" pkg="spatio_temporal_voxel_layer" output="screen">

    <rosparam file="$(find spatio_temporal_voxel_layer)/example/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find spatio_temporal_voxel_layer)/example/standard_indoor_environment_config.yaml" command="load" ns="global_costmap"/>
    </node>
</launch>

Thank you.

tekrajchhetri, 2020-03-03:
Hi,
I did understand that it was only for tests and was not included. But the thing is it keeps on looking at it and producing the error. Below is the launch file I used and the corresponding generated an error. 

Also, you specified to use STVL in move base but I did not find any information about it in the readme. Could you point to some documentation that describes on using it?

[image]

[image]

Thanks

SteveMacenski, 2020-03-03:
Please review tutorials on costmap plug-ins.

closed 2020-03-30 by SteveMacenski