SteveMacenski/spatio_temporal_voxel_layer #162: Local Costmap Update Problem with Realsense D435

hngu97, 2020-03-30:
Hi,
I have problem using this package in ROS Melodic with Realsense D435. The PointCloud2 publishing rate is around 10 - 15 Hz. Without this plugin (only laser and inflation), the updated speed of the local costmap is at least 8Hz. However, when using this plugin, the updated speed is very slow, and sometimes throw me some errors. 
I only use this plugin for local costmap. The local costmap params are:

```
local_costmap:
  global_frame: odom
  robot_base_frame: base_link
  robot_radius: 0.25
  update_frequency: 8.0
  publish_frequency: 2.0
  static_map: false
  rolling_window: true
  width: 5.0
  height: 5.0
  origin_x: 0.0
  origin_y: 0.0
  resolution: 0.05
  transform_tolerance: 0.2

  plugins:
  - name: obstacles
    type: "costmap_2d::ObstacleLayer"
  - name: spatio_temporal_voxel_layer
    type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"
  - name: inflation
    type: "costmap_2d::InflationLayer"

  spatio_temporal_voxel_layer:
      enabled:                  true
      voxel_decay:              3     #seconds if linear, e^n if exponential
      decay_model:              0      #0=linear, 1=exponential, -1=persistent
      voxel_size:               0.05   #meters
      track_unknown_space:      false   #default space is unknown
      max_obstacle_height:      2.0   #meters
      unknown_threshold:        15     #voxel height
      mark_threshold:           0      #voxel height
      update_footprint_enabled: true
      combination_method:       1      #1=max, 0=override
      obstacle_range:           3.0    #meters
      origin_z:                 0.0    #meters
      publish_voxel_map:        true   # default off
      transform_tolerance:      0.3    # seconds
      mapping_mode:             false  # default off, saves map not for navigation
      map_save_duration:        60     #default 60s, how often to autosave
      observation_sources:      depth_tracking
      depth_tracking:
        enabled: true
        data_type: PointCloud2
        topic: camera/depth/color/points
        #topic: camera/depth/points
        marking: true
        clearing: true
        min_z: 0.1                   #default 0, meters
        max_z: 3.5                   #default 10, meters
        vertical_fov_angle: 0.9
        horizontal_fov_angle: 1.5
        min_obstacle_height: 0.08     #default 0, meters
        max_obstacle_height: 1.8     #default 3, meters
        expected_update_rate: 5.0    #default 0, if not updating at this rate at least, remove from buffer
        observation_persistence: 0.0 #default 0, use all measurements taken during now-value, 0=latest
        inf_is_valid: false          #default false, for laser scans
        clear_after_reading: true    #default false, clear the buffer after the layer gets readings from it
        voxel_filter: true           #default off, apply voxel filter to sensor, recommend on
        voxel_min_points: 0
        decay_acceleration: 1.       #default 0, 1/s^2. If laser scanner MUST be 0
        model_type: 0
```
Please look at the video for more information:
[Video 1](https://www.youtube.com/watch?v=TRB5k8lK8fo&list=PL4s5Y2oPuXf8bb3DD5XAl20RFi7KTtcN3&index=4&t=0s)

Thank you.

SteveMacenski, 2020-03-30:
Try turning off voxel map publishing. That’s only for debug purposes.

hngu97, 2020-04-01:
Hi, thanks for the help. Unfortunately, it does not improve the performance. Seems like the PointCloud2 size is too large (the depth image is 848*480), so I have to greatly downsample it. It looks ok now (although it's not always meet the desired update rate).

SteveMacenski, 2020-04-01:
What’s the platform? That is a reasonably large depth image, downsampling that seems advisable.

SteveMacenski, 2020-04-02:
The `voxel_filter` also takes some CPU, maybe try turning that off. There's a trade off that makes that either it'll help or hurt but its non-linear.

hngu97, 2020-04-05:
Thanks a lot. It's better now. I can get the updated speed around 5 - 15Hz for local costmap. However, I still have one problem. I'm not sure why but it takes around 2s for marking and clearing the costmap (even though move_base doesn't give any warning or error). I get the same behaviour if I use the costmap_2d::VoxelLayer instead of this layer. I don't have this problem with laser and inflation layer. Is it normal behaviour or just something wrong with my laptop? How can I get the update speed like the robot in the supermarket? 
The PointCloud2 is filtered with a leaf size of 0.2m. My current local_costmap params are as follow:
```
local_costmap:
  global_frame: odom
  robot_base_frame: base_link
  #map_type: voxel
  robot_radius: 0.25
  update_frequency: 8.0
  publish_frequency: 1.0
  static_map: false
  rolling_window: true
  width: 5.0
  height: 5.0
  origin_x: 0.0
  origin_y: 0.0
  resolution: 0.05
  transform_tolerance: 0.2

  plugins:
  - name: obstacles
    type: "costmap_2d::ObstacleLayer"
  - name: spatio_temporal_voxel_layer
    type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"
  - name: inflation
    type: "costmap_2d::InflationLayer"

  spatio_temporal_voxel_layer:
      enabled:                  true
      voxel_decay:              3     #seconds if linear, e^n if exponential
      decay_model:              0      #0=linear, 1=exponential, -1=persistent
      voxel_size:               0.05  #meters
      track_unknown_space:      false   #default space is unknown
      max_obstacle_height:      1.8   #meters
      unknown_threshold:        15     #voxel height
      mark_threshold:           0      #voxel height
      update_footprint_enabled: true
      combination_method:       1      #1=max, 0=override
      obstacle_range:           2.5    #meters
      origin_z:                 0.0    #meters
      publish_voxel_map:        false   # default off
      transform_tolerance:      0.3    # seconds
      mapping_mode:             false  # default off, saves map not for navigation
      map_save_duration:        60     #default 60s, how often to autosave
      observation_sources:      depth_tracking
      depth_tracking:
        enabled: true
        data_type: PointCloud2
        #topic: camera/depth/color/points
        topic: filtered_points
        marking: true
        clearing: true
        min_z: 0.02                   #default 0, meters
        max_z: 2.5                   #default 10, meters
        vertical_fov_angle: 0.9
        horizontal_fov_angle: 1.5
        min_obstacle_height: 0.05     #default 0, meters
        max_obstacle_height: 1.8     #default 3, meters
        expected_update_rate: 0    #default 0, if not updating at this rate at least, remove from buffer
        observation_persistence: 0 #default 0, use all measurements taken during now-value, 0=latest
        inf_is_valid: false          #default false, for laser scans
        clear_after_reading: true    #default false, clear the buffer after the layer gets readings from it
        voxel_filter: false          #default off, apply voxel filter to sensor, recommend on
        voxel_min_points: 0
        decay_acceleration: 1.       #default 0, 1/s^2. If laser scanner MUST be 0
        model_type: 0

```

SteveMacenski, 2020-04-05:
I don’t think it takesv2s. I think youre looking in rviz which is proportional to the publish rate you have set to 1s.

closed 2020-04-07 by hngu97