SteveMacenski/spatio_temporal_voxel_layer #163: Tiny improvement

facontidavide, 2020-04-06:
Hi,

if I don't mistake, you can safely remove this line of code, since you don't care about the magnitude of the vector but also if the cross product it is positive or not.

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/eloquent-devel/src/frustum_models/depth_camera_frustum.cpp#L179

Cheers

Davide

SteveMacenski, 2020-04-06:
This is true - it could be removed. Want to submit a PR to comment it out? (I don't want removed entirely in case I need to go back and do some debug).

facontidavide, 2020-04-06:
Be my guest

closed 2020-04-06 by facontidavide

SteveMacenski, 2020-04-06:
@facontidavide actually just thinking about it - I think it does matter. 

If for instance, the p_delta is very elogated (far away) in positive X or negative Y, then the signs would get messed up since the dot product is adding them together. A very negative number + a very (and non-linearly related) positive number can have different signs depending on the magnitude

Since X and Y aren't related and you don't know all the coordinates being thrown in here, by not making it a unit vector that could change the sign

facontidavide, 2020-04-06:
If I haven't forgotten algebra, I think you are wrong 😜 

https://i.ytimg.com/vi/ui-Pc7-wFko/maxresdefault.jpg

Try with any vector and you will see.