SteveMacenski/spatio_temporal_voxel_layer #164: Is there an error in the minimum-maximum frustum distance?

facontidavide, 2020-04-06:
Hi,

when I was inspecting the code, I found something quite suspicious.

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/eloquent-devel/src/frustum_models/depth_camera_frustum.cpp#L97-L98

According to this mathematical formula, **min** and **max** distance refer to the distance between the origin and the eight corners of the frustum.

But I think this is NOT what you wanted.

If I don't mistake, **_min_d** and **_max_d** should refer to the distance of the planes parallel to the XY plane and perpendicular to the Z one.

facontidavide, 2020-04-06:
Right now I am super busy, so I don't have  time to prepare a proper pull request, but I refactored from the ground up the depth camera frustum and the code is much easier to read.

Feel free to use this if you want:

https://bitbucket.org/snippets/df-bor/9nM868/frustum-calculation-fast-and-simple

SteveMacenski, 2020-04-06:
I think you are correct, it should be  max (or min) * cos(phi/2) by that math, but it is a little convoluted.

facontidavide, 2020-04-06:
I wanted to use your frustum code, when I realized the very same math could be written in a more simple way 😅

SteveMacenski, 2020-04-06:
> I wanted to use your frustum code, when I realized the very same math could be written in a more simple way 

Yeah. I wasn't too concerned with the complexity on this since it is cached on startup. Once it worked, it worked, and I moved on. I was also trying to use the openVDB tools since I was learning that at the same time. It could be simplified for sure.

facontidavide, 2020-04-06:
I love this library, you did a great job... And you game me an excuse to learn OpenVDB 🤗

SteveMacenski, 2020-04-06:
I barely touched the surface of what its really good at. If you actually set it up as a signed distance field, you can get all the raycasting speeds ups too. However, I really never figured out a way to do that reliably with "real" data from "real" sensors since the surface of obstacles with noisy sensors isnt going to be consistent. Maybe it doesn't matter if I guestimate the surface but the overall the raycasting without the signed distance speed ups weren't enough for my needs. I needed something to drop CPU a bunch so I could process 8+ depth cameras. 

If I were to start from scratch, I'd probably take another crack at it.

facontidavide, 2020-04-07:
> Yeah. I wasn't too concerned with the complexity on this since it is cached on startup.

Of course it is not about CPU, it is just for readability purposes. If the code is readable, it is easier to find bugs or fault in the reasoning... like min/max distance !

SteveMacenski, 2021-03-10:
Resolving here shortly for Noetic and Foxy + newer in ROS2.

closed 2021-03-10 by SteveMacenski