SteveMacenski/spatio_temporal_voxel_layer #166: Spatio Temporal Layer with LaserSacan topic

AlessandroMelino, 2020-05-08:
Hello.

I have a Realsense camera to detect obstacles with pointcloud and a RPLidar to detect obstacles with laser. 

I had problems with VoxelLayer and ObstacleLayer. They mark every obstacle and they dont clear the obstacles the sensor was not seeing anymore, so the map keep full of marks.

I solve the problem using Spatio Temporal Voxel Layer to generate ostacles with pointcloud, but there is any way to use it for the LaserScan information? I just want to keep the mark of the obstacles the sensor is reading, and clear them when the sensor dont detect them, as the Spatio Temporal Voxel Layer does. Any ideas?

Thanks in advance.

Best regards.
Alessandro

nickvaras, 2020-05-08:
Yes. Even though Steve does not recommend using STVL for 2D lasercans in the Readme, I've found it to work perfectly fine with a single marking cloud with short persistence (no clearing frustum), and a good alternative to the ObstacleLayer if you want temporal decay instead of raytracing for clearing obstacles.

SteveMacenski, 2020-05-08:
> there is any way to use it for the LaserScan information

Yes, I suppose a LaserScan type but like @nickvaras (nick-o-varas, always mix between the two :wink:) mentions, I don't particularly recommend it. Typically lidars are dense enough that the raycasting is really good and fast. I work with professional grade lidars so I can't speak for the RPLidar, but proportional to range, I suspect the ray casting is good enough. 

At the end of the day, it only takes an hour to sit down and test them both out to make a decision.

AlessandroMelino, 2020-05-11:
Thank you for your answer @nickvaras  and @SteveMacenski.

As you said nickvaras, It is possible setting just one marking cloud with short persistance, but I don't know how to do it.

What I have to modify from the following code example:

```
rgbd_obstacle_layer:
  enabled:               true
  voxel_decay:           20     #seconds if linear, e^n if exponential
  decay_model:           0      #0=linear, 1=exponential, -1=persistent
  voxel_size:            0.05   #meters
  track_unknown_space:   true   #default space is unknown
  observation_persistence: 0.0  #seconds
  max_obstacle_height:   2.0    #meters
  unknown_threshold:     15     #voxel height
  mark_threshold:        0      #voxel height
  update_footprint_enabled: true
  combination_method:    1      #1=max, 0=override
  obstacle_range:        3.0    #meters
  origin_z:              0.0    #meters
  publish_voxel_map:     true   # default off
  transform_tolerance:   0.2    # seconds
  mapping_mode:          false  # default off, saves map not for navigation
  map_save_duration:     60     #default 60s, how often to autosave
  observation_sources:   rgbd1_clear rgbd1_mark
  rgbd1_mark:
    data_type: PointCloud2
    topic: camera1/depth/points
    marking: true
    clearing: false
    min_obstacle_height: 0.3     #default 0, meters
    max_obstacle_height: 2.0     #defaule 3, meters
    expected_update_rate: 0.0    #default 0, if not updating at this rate at least, remove from buffer
    observation_persistence: 0.0 #default 0, use all measurements taken during now-value, 0=latest 
    inf_is_valid: false          #default false, for laser scans
    clear_after_reading: true    #default false, clear the buffer after the layer gets readings from it
    voxel_filter: true           #default off, apply voxel filter to sensor, recommend on 
    voxel_min_points: 0          #default 0, minimum points per voxel for voxel filter
  rgbd1_clear:
    enabled: true                #default true, can be toggled on/off with associated service call
    data_type: PointCloud2
    topic: camera1/depth/points
    marking: false
    clearing: true
    min_z: 0.1                   #default 0, meters
    max_z: 7.0                   #default 10, meters
    vertical_fov_angle: 0.7      #default 0.7, radians
    horizontal_fov_angle: 1.04   #default 1.04, radians
    decay_acceleration: 1.       #default 0, 1/s^2. If laser scanner MUST be 0
    model_type: 0                #default 0 (depth camera). Use 1 for 3D Lidar
```

Sorry for this questions... But I am a bit new on this topics and I am quite lost.

Thank you for your help.

Best regards.
Alessandro

nickvaras, 2020-05-11:
Just get rid of the entire rgbd1_clear chunk, and set the voxel decay to a short time, say, 0.5 seconds or so. You might also want to look at  [](http://wiki.ros.org/nonpersistent_voxel_grid) for this type of thing.

SteveMacenski, 2020-05-11:
Please read the docs which describe the params. I offer several example configs in the repo as well. 

For config questions please ask on ROS Answers

closed 2020-05-11 by SteveMacenski