SteveMacenski/spatio_temporal_voxel_layer #167: Noetic release ?

doisyg, 2020-05-27:
Builds fine under Noetic from branch melodic-devel but at execution I get:

> terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
>   what():  Failed to load library /home/ws/devel/lib//libspatio_temporal_voxel_layer.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /lib/x86_64-linux-gnu/libjemalloc.so.2: cannot allocate memory in static TLS block)
> 

Did somebody tried already and have a different result ?

SteveMacenski, 2020-05-27:
I'm not seeing anything here that makes me suspect of pluginlib http://wiki.ros.org/noetic/Migration. The melodic devel branch looks identical to the noetic devel branch (https://github.com/ros/pluginlib). 

Does the rospack get plugins call detect it properly?

I'd be more than happy to release it once we work through these issues.

doisyg, 2020-05-28:
Yes:
```
$ rospack plugins --attrib=plugin costmap_2d 
spatio_temporal_voxel_layer /ws/src/spatio_temporal_voxel_layer/costmap_plugins.xml
costmap_2d /opt/ros/noetic/share/costmap_2d/costmap_plugins.xml
```
What's strange is that when preloading jemalloc with `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libjemalloc.so.2`, the error disapears.
I will keep you posted if I have time to investigate

SteveMacenski, 2020-05-28:
Running gdb to know where the crash happens would be useful. Might even be on movebase side if it crashes after the “complete” message

doisyg, 2020-05-29:
I assume you speak about the second issue, it looks like it is from the `updateFootprint` function:

```
[ INFO] [/move_base]: local_costmap/rgbd_obstacle_layer created underlying voxel grid.
[ INFO] [/move_base]: local_costmap/rgbd_obstacle_layer initialization complete!
[New Thread 0x7fffe65ce700 (LWP 69227)]
--Type <RET> for more, q to quit, c to continue without paging--

Thread 11 "move_base" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe65ce700 (LWP 69227)]
0x00007ffff1eaba97 in spatio_temporal_voxel_layer::SpatioTemporalVoxelLayer::updateFootprint(double, double, double, double*, double*, double*, double*) () from /ws/devel/lib//libspatio_temporal_voxel_layer.so

```
However I don't know if this second issue is linked or not to the first one and its workaround (preloading jemalloc)

doisyg, 2020-05-29:
If started with stvl param `enabled: false`  or `update_footprint_enabled : false` (and still preloading jemalloc), it doesn't crash. 
Then, at runtime (changing dyn param):
If `enabled: true` and `update_footprint_enabled : false`, no crash
Then if `enabled: true` and `update_footprint_enabled : true` => crash

SteveMacenski, 2020-05-29:
What was the crash from it you had GDB up? What was the traceback? Maybe your footprint parameter wasn't read in correctly so one of the double pointers was null?

SteveMacenski, 2020-05-29:
It would be good to know where the error is in this function https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/spatio_temporal_voxel_layer.cpp#L477-L496

(since some of them are costmap calls, then maybe actually an error in costmap 2d)

doisyg, 2020-06-01:
I am not fully understanding why, but because the updateFootprint function has no return value, the for loop was running way above `_transformed_footprint.size()` and overflowing
https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/67a6e4d93e71168a2da068c374c96bdc83c29e4a/src/spatio_temporal_voxel_layer.cpp#L477-L496
Fixed in https://github.com/SteveMacenski/spatio_temporal_voxel_layer/pull/168

SteveMacenski, 2020-06-01:
Weird. Is noetic otherwise working?

doisyg, 2020-06-01:
Yes, provided that this line is added to .bashrc `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libjemalloc.so.2`
I don't think it can be released until this issue is solved. It has apparently to do with the new jemalloc version in Focal: https://github.com/jemalloc/jemalloc/issues/1237

SteveMacenski, 2020-06-01:
is that a pluginlib issue? Is there a ticket filed for that so someone knows its an issue?

doisyg, 2020-06-01:
I found nothing else. How could you tell if it is pluginlib or not related ?

SteveMacenski, 2020-06-01:
You said you're having issues loading plugins, unless you're saying that this is a unique issue to STVL. I have to assume your issues are the result of pluginlib issues.

doisyg, 2020-06-01:
Sorry if I was unclear,  the only thing I know for sure is that when I start move_base with STVL, i get this crash:
```
[ INFO] [/move_base]: local_costmap: Using plugin "obstacle_layer"
[ INFO] [/move_base]:     Subscribed to Topics: laser_scan_sensor
[ INFO] [/move_base]: local_costmap: Using plugin "sonar_layer"
[ INFO] [/move_base]: local_costmap/sonar_layer: ALL as input_sensor_type given
[ INFO] [/move_base]: RangeSensorLayer: subscribed to topic /sonar
[ INFO] [/move_base]: local_costmap: Using plugin "rgbd_obstacle_layer"
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  Failed to load library /home/gd/elodie1_ws/devel/lib//libspatio_temporal_voxel_layer.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /lib/x86_64-linux-gnu/libjemalloc.so.2: cannot allocate memory in static TLS block)
```
whereas without it, I don't have this issue (and other plugins like `costmap_2d::StaticLayer`, `costmap_2d::ObstacleLayer`, `range_sensor_layer::RangeSensorLayer` are loading fine).
Now, if it is a STVL or a pluginlib issue, I have no clue.
I dug a bit and found out that the issue disappears by using `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libjemalloc.so.2` before launching move_base, that may help diagnostic the issue for somebody familiar with the dynamic library and memory allocation system (which I am not)

SteveMacenski, 2020-06-01:
What's the computer you're trying to run it on? Can you verify on another hardware machine it works or doesn't?

SteveMacenski, 2020-06-01:
I'd also be curious if you ran into issues loading NPVL or if its just STVL https://github.com/SteveMacenski/nonpersistent_voxel_layer

doisyg, 2020-06-01:
> What's the computer you're trying to run it on? Can you verify on another hardware machine it works or doesn't? I'm not sure what to do about that myself.

Same result on my 3 years old asus laptop, a NUC8 and a NUC10.

doisyg, 2020-06-01:
> I'd also be curious if you ran into issues loading NPVL or if its just STVL https://github.com/SteveMacenski/nonpersistent_voxel_layer

I ll try

SteveMacenski, 2020-06-01:
Ok, so probably not platform specific. 

It may (?) be released, but rosdistro doesn't have a Focal entry for openVDB https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3005

doisyg, 2020-06-01:
> > I'd also be curious if you ran into issues loading NPVL or if its just STVL https://github.com/SteveMacenski/nonpersistent_voxel_layer
> 
> I ll try

No issue with NPVL

SteveMacenski, 2020-06-01:
Mhm, that is interesting then. I don't have a 20.04 machine yet to try to debug this. Hopefully in the next few weeks but for the moment, there's not much I can do.

doisyg, 2020-06-01:
> Ok, so probably not platform specific.
> 
> It may (?) be released, but rosdistro doesn't have a Focal entry for openVDB https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3005

https://github.com/ros/rosdistro/pull/25257

doisyg, 2020-06-01:
> Mhm, that is interesting then. I don't have a 20.04 machine yet to try to debug this. Hopefully in the next few weeks but for the moment, there's not much I can do.

No rush for a release, I ll test it on a real robot hopefully in the next days and report back if I find any other issues

SteveMacenski, 2020-06-01:
Ok, that issue is really odd though, I'd suggest filing a ticket for it on pluginlib repo since something broke during the last distribution update. As far as  Ican tell, nothing should need to be changed http://wiki.ros.org/noetic/Migration

SteveMacenski, 2020-06-03:
Would this not be something to file with the openvdb maintainers on GitHub since they just need to change some flags and rebuild debians?

I'm not very encouraged by the likelihood of this being fixed given that ticket hasn't had motion since Feb.

SteveMacenski, 2020-06-09:
I filed the ticket above on OpenVDB to make them aware. Is there any other action here I can take?

doisyg, 2020-06-09:
Thanks for filling the ticket, I reported there that the issue disappears when installing openvdb7.0 instead of openvdb6.2 before recompiling. The problem now being that openvdb7.0 is not officially available on focal

SteveMacenski, 2020-06-09:
OK, mhm, we'll see what they say.

SteveMacenski, 2020-07-08:
@doisyg any changes? I'm in another cycle of releases and STVL is ready to go when this issue is resolved. I haven't heard anything from the openvdb side, but I didn't know if they updated and since you've been OK

doisyg, 2020-07-09:
Still working fine for me with the `LD_PRELOAD` workaround

SteveMacenski, 2020-07-20:
https://github.com/AcademySoftwareFoundation/openvdb/issues/732#issuecomment-661294038 

Can you verify that the binaries resolve this issue for your application?

doisyg, 2020-07-20:
Sure, as soon as I will notice an update on the Ubuntu packages, I will test and report. For now, it is still version  6.2.1-8ubuntu1 from 22 Feb 2020

SteveMacenski, 2020-07-20:
As well

SteveMacenski, 2020-07-29:
Note for ROS2 users: Nav2 tutorial includes this note to export. Hopefully we get new binaries soon, but I have to say I'm really unclear who releases the binaries for openvdb and how to get them to update them

dirk-thomas, 2020-08-05:
The very same issue exists in other packages: see ros-visualization/rqt_image_view#38.

SteveMacenski, 2020-08-05:
Fun. At least we're not alone on this front. It looks like openVDB fixed things in mainline, but I don't exactly know how to get new binaries released. Until then, Focal is basically just a dead operating system to me since I can't run openVDB which STVL and much of my R&D for nav2 environmental modeling is based on...

It looks like from @doisyg's comments that its reflected in v7.0 (is that accurate)? If so, I could update an openvdb on rosdistro to install 7.x if those binaries are available in Focal. Beyond that, I don't know how to continue.

doisyg, 2020-08-06:
> It looks like from @doisyg's comments that its reflected in v7.0 (is that accurate)? If so, I could update an openvdb on rosdistro to install 7.x if those binaries are available in Focal. Beyond that, I don't know how to continue.

Yes, I confirm that the problem disappear when manually installing libopenvdb7.0 from groovy. But these binaries are not available on focal and my request of back-porting it has been refused, see here: https://answers.launchpad.net/ubuntu/+source/openvdb/+question/691239
"Backporting 7.0 release to focal is not option."

gavanderhoorn, 2020-08-06:
@doisyg: there is actually a good rationale for why "a backport is not an option" (from comment nr 5 [here](https://answers.launchpad.net/ubuntu/+source/openvdb/+question/691239)):

> The Backports Project is a means to provide new features to users. Because of the inherent stability risks in backporting packages, users do not get backported packages without some explicit action on their part. This generally makes backports an inappropriate avenue for fixing bugs

this makes sense to me.

doisyg, 2020-08-06:
> Fun. At least we're not alone on this front.

Interesting to see the problem affects `rqt_image_view` too but that it doesn't use openvdb. This ties it up to the way pluggins are loaded and libjemalloc.

doisyg, 2020-08-06:
> @doisyg: there is actually a good rationale for why "a backport is not an option" (from comment nr 5 [here](https://answers.launchpad.net/ubuntu/+source/openvdb/+question/691239)):
>
>
> this makes [...]

Yes, I understand the rational for not backporting, but I don't know how to trigger an SRU procedure that will use a fixed version of openvdb6.2 for focal (where this fixed version will be compiled from?)

gavanderhoorn, 2020-08-06:
>  I don't know how to trigger an SRU procedure

According to comment nr 3 (I can't seen to directly link to these comments):

> If you run: `ubuntu-bug openvdb`
> 
> This will start the process for you.

doisyg, 2020-08-06:
Yes, I even reported a bug https://bugs.launchpad.net/ubuntu/+source/jemalloc/+bug/1882998 which is confirmed (and reported fixed on groovy with v7 but no backport possible). 
The thing is, I am not sure what sources ubuntu uses for the binaries of the v6.2, I guess these: https://github.com/AcademySoftwareFoundation/openvdb/tree/v6.2 but the problem is likely to still be present in these sources

SteveMacenski, 2020-08-06:
>  I don't know how to trigger an SRU procedure that will use a fixed version of openvdb6.2 for focal (where this fixed version will be compiled from?)

Exactly. I don't care about 7.0 as much as I care that 6.x is fixed in the binaries. If its just build flags, its easy to resolve, just needs someone to physically do it and know who that person is to ask. I emailed someone on the list and never got a response. I don't know if they'd object to an update to 6.3 or something if some small code changes were required. At the moment, those binaries are basically useless to most users I have to think.

j-rivero, 2020-08-06:
Hello Doisy:

> Yes, I understand the rational for not backporting, but I don't know how to trigger an SRU procedure that will use a fixed version of openvdb6.2 for focal (where this fixed version will be compiled from?)

Speaking about how to add a source code patch to a .deb package this page can provide some hints: https://packaging.ubuntu.com/singlehtml/index.html#document-ubuntu-packaging-guide/patches-to-packages

For the whole SRU process, extensive documentation is available at:
https://wiki.ubuntu.com/StableReleaseUpdates

The initial comment should look something like:
https://bugs.launchpad.net/ubuntu/+source/urdfdom-headers/+bug/1817595/comments/6

AlessandroMelino, 2020-10-27:
Hello. 

I am trying to install the spatio-temporal-voxel-layer package from source but it is gaving me problems. 

Is going to be any release for noetic and the possibility to install it with apt-get?

If not, how do you recommend me to install it from source?

Best regards.
Alessandro

SteveMacenski, 2020-10-27:
Still waiting on a fix to the binaries for openVDB that this depends on.

AlessandroMelino, 2020-10-28:
Understood. Do you have a time estimation of how much time would it take?

SteveMacenski, 2021-03-10:
https://github.com/SteveMacenski/spatio_temporal_voxel_layer#interesting-side-note A note was left about this for Foxy + newer and Noetic in the readme. 

NOTE: If used on Ubuntu 20.04 (Foxy or Noetic), you must set your LD_PRELOAD path to include jemalloc due to a known compiler flag issue in the 20.04 binaries of OpenVDB (e.x. export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libjemalloc.so.2). If you see the error: Could not load library LoadLibrary error: /usr/lib/x86_64-linux-gnu/libjemalloc.so.2: cannot allocate memory in static TLS block, this is your issue.

You could set this in the terminal, in your bashrc, or even in a launch file call if you liked. I considered just adding a system call in C++ to call this each time on startup, but I didn't know what that would do for non-Linux based OS. 

This is not something in my direct control to fix, this is an issue with openVDB and this is a reasonable single OS solution to it. I wish they would update the binaries but its out of our control.

cosmicog, 2021-03-12:
> You could set this in the terminal, in your bashrc, or even in a launch file call if you liked.

NOTE: setting it in `.bashrc` caused my google chrome to give segfaults. 
I think it is best to set in the launch file. A little help to others:
```xml
 <launch>
    <env name="LD_PRELOAD" value="/usr/lib/x86_64-linux-gnu/libjemalloc.so.2"/>
    <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
...
```

doisyg, 2021-04-02:
> > You could set this in the terminal, in your bashrc, or even in a launch file call if you liked.
> 
> NOTE: setting it in `.bashrc` caused my google chrome to give segfaults.
> I think it is best to set in the launch file. A little help to others:
> 
> ```
>  <launch>
>     <env name="LD_PRELOAD" value="/usr/lib/x86_64-linux-gnu/libjemalloc.so.2"/>
>     <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
> ...
> ```

Thanks for the tips!
ROS2 launch equivalent: `SetEnvironmentVariable('LD_PRELOAD','/usr/lib/x86_64-linux-gnu/libjemalloc.so.2')`

SteveMacenski, 2021-04-12:
I pinned this issue for visibility, but due to the addition to the readmes and pinning, I'm going to close this issue as won't fix since we have no path to releasing for Noetic / Foxy on 20.04 due to these issues out of our control -- but with reasonable workarounds. 

I'd be happy to reopen and discuss a solution if someone has the ability to re-release openVDB or have a combination of settings / cmake changes we can make to STVL to automate this issue.

HappySamuel, 2021-11-03:
Hi 

Any noetic release of STVL (install via apt) ?

Best,
Samuel

doisyg, 2021-12-01:
Looks like it is finally going to get resolved for Ubuntu: https://bugs.launchpad.net/ubuntu/+source/openvdb/+bug/1882998
I will test and report here if nobody do it first

SteveMacenski, 2021-12-01:
Freckin' awesome! I can release binaries after this is out and about. Let me know!  I thought I filed a ticket on that too and was roundly denied (though honestly I haven't used launchpad before and I'm sure I didn't follow convention)

SteveMacenski, 2021-12-01:
Funny enough, I was just thinking about updating Nav2 to use OpenVDB as the default voxel grid implementation so that we can support arbitrary heighted robots, not being limited by 16 voxel heights. I'm still not entirely sure on the memory footprint, but would be worth a try. In the early days of STVL I had actually used OpenVDB's raytracing and replicated behavior like the Voxel Layer -- though wasn't any faster since this doesn't make use of signed distance fields. 

If we can get binaries out and working + 22.04 has the fix too, then I'd be alot more confident in starting in that direction. I see no reason why Nav2 should have its own voxel_grid implementation - there's got to be a bunch of standard library options for that these day.s

doisyg, 2021-12-05:
The focal "proposed" update package fixes the issue! I reported it and it should be pushed -updated next (I will report here when it is).

> Funny enough, I was just thinking about updating Nav2 to use OpenVDB as the default voxel grid implementation so that we can support arbitrary heighted robots, not being limited by 16 voxel heights. I'm still not entirely sure on the memory footprint, but would be worth a try. In the early days of STVL I had actually used OpenVDB's raytracing and replicated behavior like the Voxel Layer -- though wasn't any faster since this doesn't make use of signed distance fields (and still had the discrete raycasting "ghost" voxels issue that was the root of what I was trying to solve).
> 
> If we can get binaries out and working + 22.04 has the fix too, then I'd be alot more confident in starting in that direction. I see no reason why Nav2 should have its own voxel_grid implementation - there's got to be a bunch of standard library options for that these days. OpenVDB is the one I know of well so I'd start with that, unless there's other reasonable alternatives.

Great news! OpenVDB still seems to be the best choice, though @facontidavide is working on something even more efficient : https://github.com/facontidavide/Bonxai

SteveMacenski, 2021-12-06:
So you did just `sudo apt install libopenvdb*` and used those binaries alone? Can you confirm @nickovaras (sorry, I'm actually not using this right now on anything so difficult for me to test confidently). 

If so, I can release it ASAP this week. 

Reopening, since we could resolve. Then I'll announce on discourse when released.

doisyg, 2021-12-06:
> So you did just `sudo apt install libopenvdb*` and used those binaries alone? Can you confirm @nickovaras (sorry, I'm actually not using this right now on anything so difficult for me to test confidently).

No, I had to enabled "proposed" updates in Ubuntu: https://wiki.ubuntu.com/Testing/EnableProposed
Then apt update +  sudo apt install libopenvdb-dev

> If so, I can release it ASAP this week.

You will be able, once the proposed pacakage is pushed to regular updates (I guess in ~ 7 days), I ll let you know here when that's the case

SteveMacenski, 2021-12-06:
Got it, let me know!

doisyg, 2021-12-08:
libopenvdb-dev 6.2.1-8ubuntu1.1 released ! (previous was 6.2.1-8ubuntu1)

[6 more replies]

closed 2021-12-15 by SteveMacenski