SteveMacenski/spatio_temporal_voxel_layer #176: Problem with parameter already declared when trying to restart controller server

Michael-Equi, 2020-08-06:
When restarting the nav2 system with this cosmtap layer the controller server is killed and emits the following error:

```
[controller_server-3] [ERROR] [1596676033.151989476] []: Caught exception in callback for transition 10
[controller_server-3] [ERROR] [1596676033.151999489] []: Original error: parameter 'stvl_layer.observation_sources' has already been declared
[controller_server-3] [WARN] [1596676033.152020113] []: Error occurred while doing error handling.

```

This is also mentioned in  https://github.com/ros-planning/navigation2/issues/1917

Michael-Equi, 2020-08-06:
I started trying to fix this by following the other costmap layers and using the declareParameter function from the cosmtap layer class instead of the rclcpp::Node declare parameter function.

https://github.com/BytesRobotics/spatio_temporal_voxel_layer/blob/restart-fix/src/spatio_temporal_voxel_layer.cpp#L81-L120

However I end up with the following error:
```

[controller_server-3] [ERROR] [1596732159.310838953] []: Caught exception in callback for transition 10
[controller_server-3] [ERROR] [1596732159.310862674] []: Original error: ArithmeticError: Tried to initialize an affine transform from a nearly singular matrix
[controller_server-3] [WARN] [1596732159.310889749] []: Error occurred while doing error handling.
[controller_server-3] [FATAL] [1596732159.310897688] [controller_server]: Lifecycle node controller_server does not have error state implemented
[lifecycle_manager-8] [ERROR] [1596732159.311433492] [lifecycle_manager_navigation]: Failed to change state for node: controller_server
[lifecycle_manager-8] [ERROR] [1596732159.311456180] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

SteveMacenski, 2020-08-06:
Huh, not sure, that looks to be like a VDB error, maybe you need to reset the voxel_grid_ pointer before making a new unique one? The actual port for declaring seems fine. E.g. https://github.com/BytesRobotics/spatio_temporal_voxel_layer/blob/restart-fix/src/spatio_temporal_voxel_layer.cpp#L143

Try adding some logging to see where it specifically crashes, 

I fixed the static_layer side of this

Michael-Equi, 2020-08-06:
It seems to happen at the following line https://github.com/BytesRobotics/spatio_temporal_voxel_layer/blob/restart-fix/src/spatio_temporal_voxel_grid.cpp#L89

Michael-Equi, 2020-08-06:
I think it is an issue with _voxel_size being initialized to 4.94066e-324 for some reason not 0.05

SteveMacenski, 2020-08-06:
`InitializeGrid` is only called on construction, not initialization, so none of that would be called on a reconfiguration unless the plugin was destroyed. that's probably from a memory slot of the other version of the plugin its reading but since it was destroyed its been set to 0 perhaps? That's an odd situation

Michael-Equi, 2020-08-06:
I figured it out. My get parameter function calls did not have a `.` between the node name and the parameter name.

Michael-Equi, 2020-08-06:
So I fixed that issue and it looks like the restarting problem has gone away but now the layer does not want to mark the costmap. The branch with the parameters declared directly from the node interface works fine but the for some reason the edits I made prevent the layer from correctly marking the map.

Michael-Equi, 2020-08-06:
Ok for some reason I am having a very hard time with this one. Accidentally deleted the enabled_ = _enabled line. Making the PR.

SteveMacenski, 2020-08-06:
Sorry - to be clear, thing are working OK now?

Michael-Equi, 2020-08-06:
Everything is working as it should now

closed 2020-08-06 by SteveMacenski