SteveMacenski/spatio_temporal_voxel_layer #184: Adding STVL layers to both the local and global costmaps

rgreid, 2020-11-05:
Hi @SteveMacenski,

We've been using STVL successfully for a year or so. In the README.md it says:

> local/global_costmap_params.yaml
> Add this plugin to your costmap params file.
>  - {name: rgbd_obstacle_layer, type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}

Here, does "local/global_costmap_params" mean local **and/or** global?  The documentation leaves us wondering...  If we add the STVL layer to **both** the local and global costmaps, will it load two copies of the plugin? (likely yes)

Are you aware of an efficient way to copy the obstacles from a STVL layer in the local costmap to the global costmap or vice versa?  i.e. Without subscribing twice and maintaining two OpenVDB structures? 

Problem: Global planner plans through walls that only appear in the local costmap via STVL; the local planner tries to follow the global plan and gets stuck. Background: We're using a VLP-16 on Melodic and TEB for the local planner. 

Thanks!

nickvaras, 2020-11-05:
Excellent question. I understand each costmap spawns it own instance. Because it is quite efficient, the performance hit is not prohibitive, but it would certainly be awesome to share the same instance between local and global cm.

SteveMacenski, 2020-11-05:
>  If we add the STVL layer to both the local and global costmaps, will it load two copies of the plugin?

Yes, please see the architectural diagrams for your uses of costmap_2d. Local and global are independent and unrelated. 

> Are you aware of an efficient way to copy the obstacles from a STVL layer in the local costmap to the global costmap or vice versa? 

There's probably a few ways to do that, but no, no costmap layer in ROS will do this. This isn't the usual type of behavior many people desire. 

> Problem: Global planner plans through walls that only appear in the local costmap via STVL; the local planner tries to follow the global plan and gets stuck. Background: We're using a VLP-16 on Melodic and TEB for the local planner.

Sounds like you should have it in your global costmap as well so your planner takes into account the current state of the environment. This is pretty standard.

Closing since I think we answered all your questions. Feel free to ask more if you have them.

closed 2020-11-05 by SteveMacenski

rgreid, 2020-11-06:
Thanks @SteveMacenski and @nickvaras. 

We'll figure a work around for now.  Not sure if this needs documenting in readme.md? I'd do a PR but there's a few branches ;)

Disappointed to not be sharing beers at ROScon again this year ;) 

Cheers