SteveMacenski/spatio_temporal_voxel_layer #189: vlp16 dones't show the voxel

chanzw, 2021-01-13:
Hello , I have a problem. I have installed the kinetic branch successfully. The pluins has worked .I can see it in the rostopic . But in RVIZ ,I choose the topic ,and It dones,t show the voxel. I run the rostopic echo, It doens't show nothing .So, I want to know the reason.But I can get  the velodyne_points    data.

 
[image]
[image]
[image]

SteveMacenski, 2021-01-13:
Do you have voxel grid publishing enabled?

chanzw, 2021-01-14:
> Do you have voxel grid publishing enabled?

ok. I know it . Thank you.

chanzw, 2021-01-14:
Hello ,The speed of the voxel clear is too slow.AndIs this problem related to computer performance？Or I can set a certain parameters.

SteveMacenski, 2021-01-14:
That might be a question better for ROS answers with more detail.

Closing ticket since we answered the question in the ticket.

closed 2021-01-14 by SteveMacenski

chanzw, 2021-01-17:
> 
> 
> That might be a question better for ROS answers with more detail.
> 
> Closing ticket since we answered the question in the ticket.

thanks