SteveMacenski/spatio_temporal_voxel_layer #190: [ROS2] Rolling window for 3d voxel layer

SBarzz, 2021-01-13:
Hi,
Hi, I'm using stvl published **voxel grid** as a map for collision detection. I want to use it as a local map which means that I want the voxel grid not to represent my temporal environment, but only to represent my local surroundings (e.g. 1m x 1m around my robot). I want the voxel grid to move along with my robot and dropping obstacle information from the map as the robot moves too far from a given area.
Is there any recommendation on how I can implement such feature?
I know that the costmap 2d map have that option but I want that also the published **3d** voxel grid will be act the same way, just like there is decay in time I want a decay in "space".
I know that it's a configuration question but I've already asked this question in the ROS answers website but there was no answer.
Thanks

SteveMacenski, 2021-01-13:
That's not the intended use for that work, but yeah you could do that if you liked. You'll have to take the voxel grid pointcloud and run a filter over it to remove things outside of the 1m x 1m box, which should be a pretty quick filter.

SBarzz, 2021-01-13:
> That's not the intended use for that work, but yeah you could do that if you liked. You'll have to take the voxel grid pointcloud and run a filter over it to remove things outside of the 1m x 1m box, which should be a pretty quick filter.

I don't think that just a filter will help me, I wan't to remove the points completely from the data structure. The whole meaning of this filter is to prevent noise.

SteveMacenski, 2021-01-13:
That is incorrect but you're welcome to use another word for removal of points outside your ROI.

SBarzz, 2021-01-13:
> That's not the intended use for that work, but yeah you could do that if you liked. You'll have to take the voxel grid pointcloud and run a filter over it to remove things outside of the 1m x 1m box, which should be a pretty quick filter.

Maybe you know about any other voxel grid which has that option?

SteveMacenski, 2021-01-13:
I don't understand what your concern is. You can do exactly what you want in a very efficient way with the links I provided and you can always implement your own box filter trivially in a few dozen lines of code. 

---

Closing - question answered

closed 2021-01-13 by SteveMacenski