SteveMacenski/spatio_temporal_voxel_layer #192: Supporting additional voxel information

karl-schulz, 2021-03-06:
Hello,

first of all - really great package. I was wondering how much changes would have to be made to support additional information in voxels apart from "free/occupied". This could be interesing for semantic labeling, RGB information or other kind of information.

The poincloud messages already support this with the "fields" definitions, where also RGB data is stored. So input and ouput would be no problem.

As I see it now, this change would require changing the leaf data type (currently a double, right?) so something else or template type. Do you think this is doable? 

I would be intereseted in implementing these changes, however, if you see any roadblocks, I would think twice.

Thanks in advance!

SteveMacenski, 2021-03-08:
This is not something likely to be merged or implemented.

SteveMacenski, 2021-03-09:
Closing - this would be a very significant change to the package that wouldn't likely be ever merged (like if you had `N` points in a voxel, what color do you use if they disagree? and is that CPU overhead worthwhile for the custom VDB grid?)

closed 2021-03-09 by SteveMacenski