SteveMacenski/spatio_temporal_voxel_layer #195: Stereo camera Voxels are created but inflation layer not updated

charlielito, 2021-03-10:
Hi !
I am using the current version for foxy in a simulated environment with a robot with a 2D lidar and a stereocamera. For the stereo camera I am using STVL layer with the configs in https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/ros2/example/outdoor_environment_config.yaml.

The problem is that despite the Voxels being created correctly (can be visualized), the global costmap does not include them correctly. Correctly in the sense that there is no inflation layer around the Voxels so the planner does not take that into account and the path created makes collide the robot.

Here is a image of an obstacle where you can see the Voxels and the costmap but there is no inflation layer created around the object.

[image]

What am I missing?

SteveMacenski, 2021-03-11:
I see laser scans, but where's the voxels? So far, I only see info that's in the static map. 

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/ros2/example/constrained_indoor_environment_config.yaml#L28 is your robot's laser even 30cm off the ground (its not for nav2's)?

charlielito, 2021-03-11:
The voxels are in blue, you can see them in front of the robot. And yes, the laser is around 40cm off the ground.
Maybe in the following video this is more noticeable

https://user-images.githubusercontent.com/8033598/110725541-c656c000-81e5-11eb-8a49-6eefa16c0249.mp4

SteveMacenski, 2021-03-11:
I fail to see the issue in this demo - what's your config file?

charlielito, 2021-03-11:
Sorry, maybe I didn't express myself correctly or I am not understanding how this plugin is supposed to work. With the following video might clear your doubts:

https://user-images.githubusercontent.com/8033598/110802051-d1dcd200-824b-11eb-8dec-fc0532fb88dc.mp4

As you see, the STVL with the stereocamera identifies an object right ahead from the robot (that the lidar does not see because the object is below the lidar laser ray). So for far so good. The problem is when navigating to a point after the obstacle, as you see, and because there is no inflation layer around the detected object, the path planned goes just around the object which makes the robot collide with the object. AFAIK there should be a inflation layer around the object so the path planned takes into account the dimensions of the robot? 

Here is my config file in `.txt` since github does not allow `.yaml`
[nav2_params.txt](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/files/6123629/nav2_params.txt)

SteveMacenski, 2021-03-11:
>       plugins: ["voxel_layer", "inflation_layer", "rgbd_obstacle_layer"]

Please read up on the documents on costmap2D, this is a _layered_ costmap. What issue might you see if you put STVL after the inflation layer when order matters?

charlielito, 2021-03-11:
That makes completely sense. I naively thought that the order didn't matter, that the framework some how knew that the inflation layer should be the last one :sweat_smile: 
It might be obvious but for beginners (like myself) it might be helpful if the documentation on costmap2D said something about the order in the list of `plugins`.
Thanks for you fast response!

closed 2021-03-11 by charlielito

SteveMacenski, 2021-03-11:
It does - please read docs more carefully before filing tickets