SteveMacenski/spatio_temporal_voxel_layer #207: map update timeout with spatio_temporal_voxel_layer!!!

JackRen88, 2021-08-06:
Hello,SteveMacenski.
When I run the move_base navigation with spatio_temporal_voxel_layer,there is a issue that is map update timeout.
common costmap configuration：
`footprint: [[-0.205, -0.155], [-0.205, 0.155], [0.077, 0.155], [0.077, -0.155]]
#robot_radius: 0.17
footprint_padding: 0.0

map_type: costmap

always_send_full_costmap: true

#obstacle_layer:
# enabled: true
# obstacle_range: 3.0
# raytrace_range: 3.5
# inflation_radius: 0.2 #not used by the layer
# track_unknown_space: true
# combination_method: 1

# observation_sources: laser_scan_sensor #laser_scan_sensor_1 
# laser_scan_sensor: {data_type: LaserScan, topic: scan, marking: true, clearing: true}
# laser_scan_sensor_1: {data_type: LaserScan, topic: scan_1, marking: true, clearing: true}

inflation_layer:
  enabled:              false
  cost_scaling_factor:  1.0  # exponential rate at which the obstacle cost drops off (default:10) astar2.0 0.8 hybrid astar 1.0 0.1
  inflation_radius:     0.1  # max. distance from an obstacle at which costs are incurred for planning paths.

static_layer:
  enabled:              false
  map_topic:            "/map"

rgbd_obstacle_layer:
  enabled:                  true
  voxel_decay:              15    # seconds if linear, e^n if exponential
  decay_model:              0     # 0=linear, 1=exponential, -1=persistent
  voxel_size:               0.05  # meters
  track_unknown_space:      true  # default space is known
  max_obstacle_height:      2.0   # meters
  unknown_threshold:        15    # voxel height
  mark_threshold:           0     # voxel height
  update_footprint_enabled: true
  combination_method:       1     # 1=max, 0=override
  obstacle_range:           3.0   # meters
  origin_z:                 0.0   # meters
  publish_voxel_map:        false # default off
  transform_tolerance:      0.5   # seconds
  mapping_mode:             false # default off, saves map not for navigation
  map_save_duration:        60    # default 60s, how often to autosave
  observation_sources:      rgbd1_mark rgbd1_clear
  rgbd1_mark:
    data_type: PointCloud2
    topic: camera_back/depth/color/points
    marking: true
    clearing: false
    min_obstacle_height: 0.3     # default 0, meters
    max_obstacle_height: 2.0     # default 3, meters
    expected_update_rate: 0.0    # default 0, if not updating at this rate at least, remove from buffer
    observation_persistence: 0.0 # default 0, use all measurements taken during now-value, 0=latest
    inf_is_valid: false          # default false, for laser scans
    voxel_filter: false          # default off, apply voxel filter to sensor, recommend on
    voxel_min_points: 0          # default 0, minimum points per voxel for voxel filter
    clear_after_reading: true    # default false, clear the buffer after the layer gets readings from it
  rgbd1_clear:
    data_type: PointCloud2
    topic: camera_back/depth/color/points
    marking: false
    clearing: true
    max_z: 7.0                  # default 0, meters
    min_z: 0.1                  # default 10, meters
    vertical_fov_angle: 0.8745  # default 0.7, radians
    horizontal_fov_angle: 1.048 # default 1.04, radians
    decay_acceleration: 5.0     # default 0, 1/s^2. If laser scanner MUST be 0
    model_type: 0                # default 0, model type for frustum. 0=depth camera, 1=3d lidar like VLP16 or similar
`
global costmap configuration:
`global_costmap:
  global_frame: odom
  robot_base_frame: base_link
  update_frequency: 1    #5
  publish_frequency: 1  #5.001
  static_map: true
 
  transform_tolerance: 0.5
  plugins:
    - {name: static_layer,            type: "costmap_2d::StaticLayer"}
    - {name: rgbd_obstacle_layer,     type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}
    - {name: inflation_layer,         type: "costmap_2d::InflationLayer"}`
  
local costpmap configuration:
 `local_costmap:
  global_frame: odom
  robot_base_frame: base_link
  update_frequency: 1.0    #10.0
  publish_frequency: 1.0 #10.001
  static_map: false
  rolling_window: true
  width: 7
  height: 7
  resolution: 0.1
  transform_tolerance: 0.5
  
  plugins:
   - {name: static_layer,        type: "costmap_2d::StaticLayer"}
   - {name: rgbd_obstacle_layer, type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}
   - {name: inflation_layer,     type: "costmap_2d::InflationLayer"}`
   

 The error that appears is 
`[ WARN] [1628254008.675817270, 126.849000000]: Map update loop missed its desired rate of 1.0000Hz... the loop actually took 1.9340 seconds
[ WARN] [1628254008.678563403, 126.851000000]: Failed to transform the goal pose from map into the odom frame: Lookup would require extrapolation into the past.  Requested time 45.142000000 but the earliest data is at time 116.950000000, when looking up transform from frame [map] to frame [odom]
`
When I use the `obstacle layer` which is default layer in navigation,everything is ok,the problem occurs when using the spatio_temporal_voxel_layer, i think my configuration of the spatio_temporal_voxel_layer is Incorrect,how  should I  solved it ?

SteveMacenski, 2021-08-06:
Please ask on ros answers. Please stop filing superfluous tickets on projects asking for configuration help. My answer is going to continue to be the same. Config questions belong on ros answers

closed 2021-08-06 by SteveMacenski

nickvaras, 2021-08-06:
https://answers.ros.org