SteveMacenski/spatio_temporal_voxel_layer #221: Unable to locate package for ROS Noetic

mbed92, 2022-01-18:
Hi. For some reason I'm unable to install the package on ROS Noetic Docker image (ros-noetic-base):
```
E: Unable to locate package ros-noetic-spatio-temporal-voxel-layer
```
According to [ROS Index](http://repositories.ros.org/status_page/ros_noetic_default.html?q=spatio_temporal_voxel_layer) everything seems to be ok, however, [ROS repositories](http://repositories.ros.org/status_page/ros_noetic_default.html?q=spatio_temporal_voxel_layer) shows that something is missing in that package.

SteveMacenski, 2022-01-18:
It’s having some dependency  issues on some noetic platforms that I haven’t had time to resolve yet in rosdep. You can always build from source until that’s worked out!

SteveMacenski, 2023-08-08:
https://build.ros.org/job/Nbin_ufhf_uFhf__spatio_temporal_voxel_layer__ubuntu_focal_armhf__binary/ noetic binaries should be available now

closed 2023-08-08 by SteveMacenski