SteveMacenski/spatio_temporal_voxel_layer #222: Package not building (error: ‘logger_’ was not declared in this scope)

TigerPeng2, 2022-01-28:
Hello. I am on a dual-booted MacBook Pro 2017 and I'm encountering an issue where, after building and installing openvdb, ROS2 Foxy, Nav2, and colcon, when I try to colcon build the stvl package. I'm encountering the following errors.
I don't know what could be causing this, as I have sourced ROS2 before running colcon build, and it appears to be an error in getting the logger_ reference, which should be inherited from the rclcpp::Node, right?

https://pastebin.com/NAQzv28f

Is there another dependency of some sort that I missed?
I am on Ubuntu 20.04 dual booted on a Macbook Pro 2017, although I should note that I tried to perform this build process on both virtualbox and vmware fusion ubuntu 20.04 images and encountered the same problems.

SteveMacenski, 2022-01-28:
The variable "logger_" doesn't exist on Foxy https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/foxy-devel/src/spatio_temporal_voxel_layer.cpp. You probably didn't checkout the Foxy branch.

SteveMacenski, 2022-01-28:
Actually looking at your line numbers, that's definitely your issue. You must use the right distribution's code with your ROS distro.

closed 2022-01-28 by SteveMacenski