SteveMacenski/spatio_temporal_voxel_layer #235: Failed to load library /opt/ros/noetic/lib//libspatio_temporal_voxel_layer.so

MikeHallettUK, 2022-07-13:
ERROR = 
"  what():  Failed to load library /opt/ros/noetic/lib//libspatio_temporal_voxel_layer.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. 
  Error string: Could not load library (Poco exception = /opt/ros/noetic/lib//libspatio_temporal_voxel_layer.so: 
  undefined symbol: _ZN10costmap_2d12CostmapLayer9clearAreaEiiiib)"

- Installed ok = "sudo apt-get install ros-noetic-spatio-temporal-voxel-layer"  = (/ros-noetic-spatio-temporal-voxel-layer_1.4.5-1focal.20220621.182041_amd64.deb)
- library libspatio_temporal_voxel_layer.so does exist in library /opt/ros/noetic/lib//
- using your standard_indoor_environment_config.yaml: e.g. "{name: rgbd_obstacle_layer,       type: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"}"
- tried re-installing... have not yet tried to compile from code (Install from source)...

Error msq is same if using amcl NAV launch, or RTABmap launch.

? any advice please ?

Thanks

Mike

SteveMacenski, 2022-07-13:
Ask on ROS Answers

closed 2022-07-13 by SteveMacenski

MikeHallettUK, 2022-07-13:
OK. Thx = is at :

[https://answers.ros.org/question/403714/failed-to-load-library-optrosnoeticliblibspatio_temporal_voxel_layerso/](https://answers.ros.org/question/403714/failed-to-load-library-optrosnoeticliblibspatio_temporal_voxel_layerso/)

MikeHallettUK, 2022-07-18:
fyi : Now compiled ok from source, with libopenvdb6.2/focal 6.2.1-8ubuntu1.1 amd64, and:
"LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libjemalloc.so.2".

This runs ok.  So it seems it was the apt install binary that failed on Ubuntu 20.04

MikeHallettUK, 2022-08-01:
ps. broken again ... I just did a general Ubuntu Software Update... and now it fails again: even though still on libopenvdb6.2/focal 6.2.1

...