SteveMacenski/spatio_temporal_voxel_layer #236: VISUALIZE_FRUSTUM

tonynajjar, 2022-07-21:
I noticed you removed the VISUALIZE_FRUSTUM "tool" in this [commit](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/commit/a8d34416c986109e361cee889b2abcf1239acabd).

I still need to visualize my frustum to understand why voxels are disappearing so quick. Any chance in getting it back? Any existing alternatives?

tonynajjar, 2022-07-21:
Also being worked on here: https://github.com/PickNikRobotics/rviz_visual_tools/issues/181

SteveMacenski, 2022-07-21:
I forget why I removed it in the port, but I do remember doing that intentionally due to some technical issue. I'm looking at the code now and have no idea why I thought that. If you were to submit a PR to add it back, I'd be happy to merge it (Unfortunately, I'm leaving for Italy in 48 hours for a few weeks, I would not be able to get to an implementation myself in that period of time).

tonynajjar, 2022-07-21:
I found that Foxglove supports visualizing the FOV from the camera_info topic
[image]

The problem is still that STVL's frustum are not created based on the camera_info topic, but based on the parameters, which is redundant information. 

Feature request: STVL creates the Frustums from the camera_info if it exists, fallbacks to parameters.

And for what VISUALIZE_FRUSTUM accomplishes, I think this should be an rviz plugin.

Sorry I won't have time to contribute here, my after-work time is currently full with the new tricycle controller

tonynajjar, 2022-07-21:
> Feature request: STVL creates the Frustums from the camera_info if it exists, fallbacks to parameters.

Let me know if you want me to create a separate feature request for that (if you agree with it)

closed 2022-08-22 by SteveMacenski