SteveMacenski/spatio_temporal_voxel_layer #239: Support other frustum shapes

tonynajjar, 2022-07-27:
The Kinect Azure is commonly used as depth camera and its frustum is not rectangular, it's either circular or an hexagonal depending on the mode. https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#depth-camera-supported-operating-modes

It would be nice to support other shapes than rectangular

SteveMacenski, 2022-08-17:
That it mighty interesting indeed. The same logic as used in the depth camera frustum model using the 6 planes would work for hexagonal ones as well (just now with 8 planes). I'd be happy to merge something if a contribution was made to support that. 

Circular is a bit easier, that would just be checking a radius and 2 planes (or even if being that simple, just range)

tonynajjar, 2022-08-17:
We were almost going to give a shot implementing it but realized that gazebo/OGRE also doesn't support other frustum shapes (looks like they are really rare). We instead went with clipping the circular frustum to a rectangular so this is low prio for us.