SteveMacenski/spatio_temporal_voxel_layer #241: Voxel_min

Srijan221, 2022-11-04:


SteveMacenski, 2022-11-04:
What version of this software are you using and is it actually released in that distribution?

Do you know for a fact that only a single measurement is coming from that voxel?

Srijan221, 2022-11-05:
Hello Steve,
We are using Melodic-devel Ros1 Distribution and download the software from binaries.
The camera used is Intel Realsense D455 and whole plugin is run on a Jetson Xavier NX.

The plus signs are the ghost points which comes at random locations and there is no object corresponding to it in the real world. Also the STVL grid only makes a cube voxel for every plus sign which I am not sure the reason it happens.

I was just wondering can `voxel_min_points` parameter fix this plus sign issue by maybe not assigning a voxel to these ghost points?

SteveMacenski, 2022-11-05:
> Also the STVL grid only makes a cube voxel for every plus sign which I am not sure the reason it happens.

That doesn't seem related to STVL, I'm not sure what that relates to though. I've never seen anything like that before unless you've made your parameters really wacky.

Print before this line the value of the min voxel size https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/measurement_buffer.cpp#L164. If you get a non-zero number, then your gripe is with pcl. If you get a zero-value, I can look into it.

> I was just wondering can voxel_min_points parameter fix this plus sign issue by maybe not assigning a voxel to these ghost points?

I don't know what / how you managed to get those plus signs (other costmap layers after this?) so I doubt that's in any way related.

nickovaras, 2022-11-08:
This is an issue with PCL. The original Voxel Filter implementation used the original XYZ pointcloud message format and worked with the `min_points` parameter. With the transition to the PointCloud2 format, the parameter was left there exposed but it doesn't do anything. If it ever gets fixed in PCL, then it will automagically work with STVL (and would be very handy indeed).

closed 2022-11-08 by nickovaras

Srijan221, 2022-11-09:
> Print before this line the value of the min voxel size https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/measurement_buffer.cpp#L164. If you get a non-zero number, then your gripe is with pcl. If you get a zero-value, I can look into it.
Moreover, we have used the default pointcloud from the ros realsense wrapper and no other costmap layers are added in our implementation.

I tried printing, but the code doesn't seem to enter that block, it doesn't print any value (zero / non-zero), not even any static string written in that line.

Those plus signs, in my opinion are due to the lighting conditions in the room, maybe due to reflections.

I have tried using two different Realsense - Realsense D415 and D455,
Below are some errors observed with Realsense d455 - 
[image]
**As you can see in the image above, the reflection due to the sunlight causes a random voxel to appear in the map and it is marked onto the costmap with an inflation which should not happen in a general case.**

[image]
**And, ghost points appear at locations that aren't in the field of view of camera as you can see above.**

I am just looking for `voxel_min_points` to somehow remove these unwanted voxels to appear or mark in the costmap altogether.