SteveMacenski/spatio_temporal_voxel_layer #242: `voxel_min_points` parameter not working

Srijan221, 2022-11-10:
> Print before this line the value of the min voxel size https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/melodic-devel/src/measurement_buffer.cpp#L164. If you get a non-zero number, then your gripe is with pcl. If you get a zero-value, I can look into it.
Moreover, we have used the default pointcloud from the ros realsense wrapper and no other costmap layers are added in our implementation.

I tried printing, but the code doesn't seem to enter that block, it doesn't print any value (zero / non-zero), not even any static string written in that line.

Those plus signs, in my opinion are due to the lighting conditions in the room, maybe due to reflections.

I have tried using two different Realsense - Realsense D415 and D455,
Below are some errors observed with Realsense d455 - 
[image]
**As you can see in the image above, the reflection due to the sunlight causes a random voxel to appear in the map and it is marked onto the costmap with an inflation which should not happen in a general case.**

[image]
**And, ghost points appear at locations that aren't in the field of view of camera as you can see above.**

I am just looking for `voxel_min_points` to somehow remove these unwanted voxels to appear or mark in the costmap altogether.

_Originally posted by @Srijan221 in https://github.com/SteveMacenski/spatio_temporal_voxel_layer/issues/241#issuecomment-1308673467_

SteveMacenski, 2022-11-10:
I don’t see how these are STVL questions / problems. It sounds like you need to calibrate your sensors or filter them if you frequently are environments where lighting conditions result in extraneous noise.

closed 2022-11-10 by SteveMacenski

Srijan221, 2022-11-10:
Thankyou for the clarification Steve, I just want to know the use of `voxel_min_points` parameter in ROS1 Melodic version of the plugin. Why is the parameter not working and changing it's value does not help?