SteveMacenski/spatio_temporal_voxel_layer #257: PointCloud2Callback not called

dujeong, 2023-03-24:
When I checked the voxel_grid topic with rviz2, nothing came out.
So I checked through `ros2 topic echo`, it was coming out in an empty array ([ ]) 
(Actual input data(realsense depth pointcloud) came out well.)

I checked and found that stvl could not receive pointcloud data of realsense coming in through input.

As a result of checking by inserting the log as below, the callback function was suddenly not called while receiving it as soon as robot was booted.

```
/*****************************************************************************/
void SpatioTemporalVoxelLayer::PointCloud2Callback(
  sensor_msgs::msg::PointCloud2::ConstSharedPtr message,
  const std::shared_ptr<buffer::MeasurementBuffer> & buffer)
/*****************************************************************************/
{
  std::cout << "in callback" << std::endl;
  if (!buffer->IsEnabled()) {
    return;
  }
  std::cout << "callback middle" << std::endl;
  // buffer the point cloud
  buffer->Lock();
  buffer->BufferROSCloud(*message);
  buffer->Unlock();
  std::cout << "callback finish" << std::endl;
}
```

At the beginning of the boot, a few lines were printed, but after 'callback finish', it was no longer printed.

For reference, the parameter `enabled` also did not work (there is a possibility that the callback related to dynamic reconfiguration also did not work)

The important thing is that sometimes the problem comes out, but sometimes it doesn't.
For your information, the obstacle layer (`LaserScan` type) used together works well.

I thought it might be an executor problem dealing with callbacks, but for that, `LaserScan` data is being handled well.

The source version I used is the latest version as of now.

dujeong, 2023-03-27:
I am sorry for not providing enough information.

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- navigation2 Version or commit hash:
  - source build: ca8c4c5f40b6fb852e198b6507e1ca186bce84c6
- DDS implementation:
  - cyclonedds

---

- If you use the normal Voxel Layer do you see this problem?
  - when I set like `observation_sources: scan pointcloud`  in voxel layer,  costmap was applied only to scan data.

- `/scan` topic information (Type: sensor_msgs/msg/LaserScan)

  ```bash
  <<< publisher >>>
  Node name: scan_merger_1
  Node namespace: /
  Topic type: sensor_msgs/msg/LaserScan
  Endpoint type: PUBLISHER
  GID: 01.10.13.7c.3f.aa.49.d1.7b.32.14.a2.00.00.4a.03.00.00.00.00.00.00.00.00
  QoS profile:
    Reliability: BEST_EFFORT
    History (Depth): KEEP_LAST (5)
    Durability: VOLATILE
    Lifespan: Infinite
    Deadline: Infinite
    Liveliness: AUTOMATIC
    Liveliness lease duration: Infinite
  
  <<< subscriber >>>
  Node name: local_costmap
  Node namespace: /local_costmap
  Topic type: sensor_msgs/msg/LaserScan
  Endpoint type: SUBSCRIPTION
  GID: 01.10.c8.68.70.d7.71.88.26.fa.40.fc.00.00.6f.04.00.00.00.00.00.00.00.00
  QoS profile:
    Reliability: BEST_EFFORT
    History (Depth): KEEP_LAST (50)
    Durability: VOLATILE
    Lifespan: Infinite
    Deadline: Infinite
    Liveliness: AUTOMATIC
    Liveliness lease duration: Infinite
  
  ```
  

- `/realsense2_camera_1/depth/color/points` topic information

  ```bash
  Type: sensor_msgs/msg/PointCloud2
  
  <<< publisher >>>
  Node name: realsense2_camera_1
  Node namespace: /
  Topic type: sensor_msgs/msg/PointCloud2
  Endpoint type: PUBLISHER
  GID: 01.10.04.4f.8c.a2.e6.a4.8f.37.64.81.00.00.16.03.00.00.00.00.00.00.00.00
  QoS profile:
    Reliability: RELIABLE
    History (Depth): KEEP_LAST (10)
    Durability: VOLATILE
    Lifespan: Infinite
    Deadline: Infinite
    Liveliness: AUTOMATIC
    Liveliness lease duration: Infinite
    
  <<< subscriber >>>
  Node name: local_costmap
  Node namespace: /local_costmap
  Topic type: sensor_msgs/msg/PointCloud2
  Endpoint type: SUBSCRIPTION
  GID: 01.10.c8.68.70.d7.71.88.26.fa.40.fc.00.00.70.04.00.00.00.00.00.00.00.00
  QoS profile:
    Reliability: BEST_EFFORT
    History (Depth): KEEP_LAST (50)
    Durability: VOLATILE
    Lifespan: Infinite
    Deadline: Infinite
    Liveliness: AUTOMATIC
    Liveliness lease duration: Infinite
  ```

dujeong, 2023-03-27:
@SteveMacenski 
As you said, I thought it might be a network problem, so I proceeded with [DDS tuning](https://docs.ros.org/en/humble/How-To-Guides/DDS-tuning.html#cyclone-dds-tuning).

Then I succeeded.
Actually, I typed `sudo sysctl -w net.core.rmem_max=2147483647`, then succeeded!
I set it up for both remote pc and robot.

Thank you for your insightful reply.

GPrathap, 2024-01-29:
I had the same issue solution was to install cyclone dds, https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html https://github.com/SteveMacenski/spatio_temporal_voxel_layer/issues/250

GPrathap, 2024-02-08:
After some time, issues start even with cyclone dds. I need to restart it several times to get it working, 
I've set these values but it seems to still message for point clouds disappeared 

        net.core.rmem_max=5147483647
        net.core.rmem_default=5147483647

dujeong, 2024-02-12:
@GPrathap Why don't you put up a new issue if the issue persists?

GPrathap, 2024-02-13:
Sorry, I forgot to update, whenever I start the docker, I have to run `sudo sysctl -w net.core.rmem_max=2147483647` . Then it is fine. Putting in /etc/sysctl.d/10-cyclone-max.conf does not work.