SteveMacenski/spatio_temporal_voxel_layer #268: Does the STVL provides the ability to identify the obstacle infromations (height, width and distance from the robot)

hmpcjayaweera, 2023-07-31:
Does the STVL provides the ability to identify the obstacle information such as distance to obstacles from the robot, obstacle height, width and length. If yes, please can you give me some information about it. 
In addition, can we use the STVL for obstacle avoidance in NAV2. When I use the STVL plugins, it shows the 3D obstacle in Rviz2, but the robot is not avoiding it in point to point navigation (I only use STVL layer and inflation layer for the local costmap). Thank you.

SteveMacenski, 2023-07-31:
You have the coordinates of the voxels in the voxel layer; so of course you can compute the distance from your robot's pose to any particular voxel or find its height. Simply use the voxel information and your robot pose information to compute the distance or look at the `Z` height of the voxel of interest. 

Please file that on ROS Answers with alot more detail and configuration files of relevant systems.

hmpcjayaweera, 2023-07-31:
When I check the "ros2 topic info /local_costmap/voxel_grid", it it shows that "Type: sensor_msgs/msg/PointCloud2", since it is not in the voxel grid format is it possible to get the obstacle coordinate from voxel layer. thanks

SteveMacenski, 2023-07-31:
That is the voxel grid.

hmpcjayaweera, 2023-08-01:
Thank you very much for your quick reply, Please can I know the format of this voxel grid (sensor_msgs/msg/PointCloud2), because these are just pixel values. How can i convert this point2 data to world coordinate. Thank you.