SteveMacenski/spatio_temporal_voxel_layer #276: Prevent navigation when the expected rate of a source is not respected

tuxnese, 2023-11-15:
When using stvl on humble I have experienced that the robot keeps navigating even if a source is not streaming despite the warnings concerning the expected rate of the sensor. I think that it would be much safer to prevent navigation in this case.
Is there already a way to do it?

SteveMacenski, 2023-11-15:
The code that I'm looking at makes me think that it shuold work properly if you set your expected update rates properly. 

We get the `current` status if we're updating correctly

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/6ade4fe4164576d4356ca92b1cd258c71930b1f4/src/spatio_temporal_voxel_layer.cpp#L482

Then set it globally 

https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/6ade4fe4164576d4356ca92b1cd258c71930b1f4/src/spatio_temporal_voxel_layer.cpp#L762

The algorithm will wait until current before enabling a new planning/control request in the main servers 

https://github.com/ros-planning/navigation2/blob/main/nav2_planner/src/planner_server.cpp#L293-L297

See a bug here I'm missing?

tuxnese, 2023-11-16:
I think there is in fact a bug. In the first function the current value is
set to the last value. In the loop the assignment should be made by using
an and operator between the new and current value. May be you could
alternatively break the loop when current gets false.
Am I right?

SteveMacenski, 2023-11-16:
True. Please test that and if it works now, please open a PR for that change inside of `GetClearingObservations` and `GetMarkingObservations`

closed 2024-03-06 by SteveMacenski