SteveMacenski/spatio_temporal_voxel_layer #279: Correct branch for ROS 2 Humble

Cristian-wp, 2023-11-23:
Hello, I am following  [(STVL) Using an External Costmap Plugin turorial](https://navigation.ros.org/tutorials/docs/navigation2_with_stvl.html#install-stvl) about using this layer in the [Nav2 ROS 2 package](https://github.com/ros-planning/navigation2/tree/humble).
I can not understand If I am doing something wrong, or the instructions does not match the actual branch names.
The instruction are:
STVL can be installed in ROS 2 via the ROS Build Farm:
```
sudo apt install ros-<ros2-distro>-spatio-temporal-voxel-layer
```
_My output:_
```
sudo apt install ros-humble-spatio-temporal-voxel-layer
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-spatio-temporal-voxel-layer

```
It can also be built from source by cloning the repository into your Navigation2 workspace:
```
git clone -b <ros2-distro>-devel git@github.com:stevemacenski/spatio_temporal_voxel_layer
```
_My output:_
```
git clone --recursive git@github.com:stevemacenski/spatio_temporal_voxel_layer -b humble-devel
Cloning into 'spatio_temporal_voxel_layer'...
The authenticity of host 'github.com (140.82.121.4)' can't be established.
ED25519 key fingerprint is SHA256:+DiY3wvvV6TuJJhbpZisF/zLDA0zPMSvHdkr4UvCOqU.
This key is not known by any other names
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

I have check the repository and I use the ros2 branch and it compiles without problem with colcon.
So I wish to know if I am not founding the repository branch `humble-devel` for some reason, or the correct one is `ros2`.
Thanks in advance.

SteveMacenski, 2023-11-23:
Use `humble` branch, not `humble-devel`, you even pointed _me_ to the `humble` branch :wink: 

Note the thread here if on 22.04: https://github.com/SteveMacenski/spatio_temporal_voxel_layer/issues/167 when you get to compiling

closed 2023-11-23 by SteveMacenski

Cristian-wp, 2023-11-24:
I mean for [spatio_temporal_voxel_layer](https://github.com/SteveMacenski/spatio_temporal_voxel_layer) not Navigation2