SteveMacenski/spatio_temporal_voxel_layer #289: 从来从

baoqianiii, 2024-04-30:
Could not find a package configuration file provided by
  "ament_cmake_vendor_package" with any of the following names:

    ament_cmake_vendor_packageConfig.cmake
    ament_cmake_vendor_package-config.cmake

SteveMacenski, 2024-04-30:
Did you source your ROS distribution before trying to build? I can see this sometimes when I build without sourcing ROS, then source it once I notice. Nuke install/build and open a new terminal, source ROS, and try again :-)

baoqianiii, 2024-05-01:
> Did you source your ROS distribution before trying to build? I can see this sometimes when I build without sourcing ROS, then source it once I notice. Nuke install/build and open a new terminal, source ROS, and try again :-)

Thanks to the author for such a quick reply, I followed the source ros2 environment you said, but I still have the same problem, can you please tell me what causes this error?

SteveMacenski, 2024-05-01:
I'm not sure - perhaps try Robotics stack exchange or clear your workspace (build, install, log), close your terminal, open a new one, source `/opt/ros` and try again

closed 2024-05-02 by baoqianiii

CharlesAntoineParent, 2024-05-09:
Same problem under ros2 after sourcing my env. @baoqianiii How did you solve your problem ?

baoqianiii, 2024-05-09:
Sorry, I didn't fix this, I tried a lot of methods. Do you have any ideas?

baoqianiii, 2024-05-09:
> Same problem under ros2 after sourcing my env. @baoqianiii How did you solve your problem ?
Sorry, I didn't fix this, I tried a lot of methods. Do you have any ideas?

andrewjong, 2024-07-19:
@SteveMacenski I still get this error on ROS2 Humble on Ubuntu 22.04.
```
--- stderr: openvdb_vendor                                                                                                                 
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findament_cmake_vendor_package.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "ament_cmake_vendor_package", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "ament_cmake_vendor_package" with any of the following names:

    ament_cmake_vendor_packageConfig.cmake
    ament_cmake_vendor_package-config.cmake

  Add the installation prefix of "ament_cmake_vendor_package" to
  CMAKE_PREFIX_PATH or set "ament_cmake_vendor_package_DIR" to a directory
  containing one of the above files.  If "ament_cmake_vendor_package"
  provides a separate development package or SDK, be sure it has been
  installed.
```

It seems ament_cmake_vendor_package was not added until ROS2 Irwin, which is after ROS2 Humble?

andrewjong, 2024-07-19:
Nevermind, it seems it was added to Humble but doesn't come with Humble by default. Fixed with `sudo apt install ros-humble-ament-cmake-vendor-package`