SteveMacenski/spatio_temporal_voxel_layer #292: Possible bug in operator() for marking voxels?

xaru8145, 2024-06-03:
Hi,

I noticed [L295](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/ros2/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_grid.cpp#L295) of spatio_temporal_voxel_grid cpp file 
`double z = *iter_y < 0 ? *iter_z - _voxel_size : *iter_z; `
Is this intended or shoud it be this way?
`double z = *iter_z < 0 ? *iter_z - _voxel_size : *iter_z; `

Thanks for your time!

SteveMacenski, 2024-06-04:
I'm pretty sure that's a typo. @KavenYau this is from your PR a little over a year ago. Do you agree this is just a typo that we should fix?

KavenYau, 2024-06-04:
> I'm pretty sure that's a typo. @KavenYau this is from your PR a little over a year ago. Do you agree this is just a typo that we should fix?

Yes, it's definitely a typo. :frowning:

SteveMacenski, 2024-06-04:
@xaru8145 can you submit a PR? :smil

xaru8145, 2024-06-04:
Done @SteveMacenski ! PR #293

closed 2024-06-04 by SteveMacenski