SteveMacenski/spatio_temporal_voxel_layer #296: Installation Hiccups [Humble] on Jetson

HappySamuel, 2024-07-04:
OS: Ubuntu 20.04
JETPACK: 5.1.2-b104
ROS2: Humble

After git clone the repo into `colcon_ws/src`, then `git checkout humble`, and then build the `openvdb_vendor` 
1) `cd openvdb_vendor`
2) `mkdir build`
3) `cd build`
4) `cmake ..` 
5) `make -j$(nproc)`
6) `sudo make install`

After building the `openvdb_vendor`, then return to `colcon_ws` and `colcon build --symlink-install` and get the following error.

[image]

```
--- stderr: openvdb_vendor                                           
CMake Error at /usr/local/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find TBB: Found unsuitable version "2020.1", but required is at
  least "2020.2" (found /usr/include, found components: tbb)
Call Stack (most recent call first):
  /usr/local/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:592 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindTBB.cmake:368 (find_package_handle_standard_args)
  openvdb/openvdb/CMakeLists.txt:133 (find_package)
```
Any idea how shall i install TBB 2020.2 on ubuntu 20.04 ?

Best
Samuel

SteveMacenski, 2024-07-04:
Humble supports 22.04, not 20.04, so I’m not sure. This is kind of the wild west if you want to use unsupported configurations unfortunately! 

Docker might help here, but also note that openvdb_vendor required *alot* of memory to compile and I’m dubious that you could do it in the first place. Why not use docker + binaries instead?

HappySamuel, 2024-07-04:
Found a solution for this TBB version issue, especially for Humble on Jetson Devices (Ubuntu 20.04)

[libtbb2  version: 2020.3-1](http://ftp.de.debian.org/debian/pool/main/t/tbb/libtbb2_2020.3-1_arm64.deb)
[libtbb-dev  version: 2020.3-1](http://ftp.de.debian.org/debian/pool/main/t/tbb/libtbb-dev_2020.3-1_arm64.deb)
`sudo dpkg -i libtbb2_2020.3-1_arm64.deb`
`sudo dpkg -i libtbb-dev_2020.3-1_arm64.deb`

After upgrading the TBB version to `2020.3-1`, then proceed to `colcon_ws` and `colcon build --symlink-install `

Best,
Samuel

closed 2024-07-04 by HappySamuel