SteveMacenski/spatio_temporal_voxel_layer #297: Callback group not set for PublisherOptions

ivrolan, 2024-07-15:
In `ros2` and `humble` branch, in `spatio_temporal_voxel_layer.cpp`, the subscriber and publisher options are set in [this way](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/ros2/spatio_temporal_voxel_layer/src/spatio_temporal_voxel_layer.cpp#L143-L147):

    auto sub_opt = rclcpp::SubscriptionOptions();
    sub_opt.callback_group = callback_group_;

    auto pub_opt = rclcpp::PublisherOptions();
    sub_opt.callback_group = callback_group_;

Note how the `sub_opt.callback_group = callback_group_` is set twice, and the `pub_opt` is left as the default. Is this intended?

SteveMacenski, 2024-07-15:
Looks like a typo to me, this isn't a big deal but certainly would be a good PR to have fixed! Can you submit that @ivrolan?

ivrolan, 2024-07-17:
Sure! I'll test it and open the PR

PR #298 (2024-07-17): Fix pub_options callback group not set
This PR fixes #297 and sets the `callback_group` of the publisher options for the `voxel_grid` publisher. Before the `callback_group` for the subscriber options was set twice.

Tested on humble with no perceptible changes, everything seems working fine. I don't know exactly the consequences of not setting the callback_group (e.g. `get_subscription_count()` was also working even with the default options without setting it), but it makes more sense to fix this typo now and prevent future problems.