SteveMacenski/spatio_temporal_voxel_layer #302: Difference between min/max_obstacle_height vs min/max_z ??

HappySamuel, 2024-08-14:
Hi

I am curious what are the difference between these parameters `min/max_obstacle_height` and `min/max_z` ??? Are there any difference for these parameters when using `3d lidar` vs `rgbd camera`

Best
Samuel

SteveMacenski, 2024-08-14:
Height is the height off of the map frame and min/max z are ranges - since usually that's the way the pointclouds from depth cameras and lidars are oriented. I can appreciate the confusion though, I could have made that named more clearly. 

A layer can have both marking and clearing, I forget why off hand we have them separated in the default examples. I think I might have the settings slightly different (or could be a simplified version of a file I was using at a company at the time I stripped back to just 1 sensor)

HappySamuel, 2024-08-15:
Hi @SteveMacenski 

Thanks for the explanation about those parameters.

Is that mean we can `marking: true` & `clearing: true` in same layer for `observation_sources`?  For example like below? Or is there anything i need to change / take note? As i am using 3D lidar.
```
        lslidar:
          data_type: PointCloud2
          topic: /lslidar_c16/points
          marking: true
          clearing: true
          min_obstacle_height: -0.2 #[m]
          max_obstacle_height: 0.5  #[m]
          expected_update_rate: 0.0 #default 0, if not updating at least this rate, remove from buffer
          observation_persistence: 0.0 #default 0, use all measurements taken during now value, 0=latest
          inf_is_valid: false  #default false, for laserscan
          voxel_filter: true   #default off, apply voxel filter to sensor, recommend on
          voxel_min_points: 20 #default 0, min pts per voxel for voxel filter
          clear_after_reading: true #default false, clear the buffer after the layer gets readings from it
          max_z: 1.0 #[m]
          min_z: -0.5 #[m]
          vertical_fov_angle: 0.5236 #[rad]
          horizontal_fov_angle: 6.2832 #[rad]
          decay_acceleration: 10.0 #default 0, 1/s^2. If laserscan MUST be 0
          model_type: 1 #model type for frustum. 0=depth_camera , 1=3d_lidar
```

Best,
Samuel

closed 2024-08-15 by SteveMacenski