SteveMacenski/spatio_temporal_voxel_layer #305: Binaries for ROS 2 Jazzy

agoeckner, 2024-10-30:
It seems that CI for ROS 2 Jazzy isn't working/enabled. The `ros-jazzy-spatio-temporal-voxel-layer` package is not found by apt on my arm64 system.

agoeckner, 2024-11-01:
I wonder whether @SteveMacenski you need to enable something in the CI system after fixing #303.

SteveMacenski, 2024-11-04:
Back from ROSCon / ROS-I in Europe now. Sorry for the delay. 

The Jazzy ARM job is failing with 
```
23:25:39 -- Found openvdb_vendor: 2.5.2 (/opt/ros/jazzy/share/openvdb_vendor/cmake)
23:25:39 -- CMAKE_BUILD_TYPE: None
23:25:39 CMake Error at /opt/ros/jazzy/share/openvdb_vendor/cmake/openvdb_vendor-extras.cmake:8 (find_package):
23:25:39   By not providing "FindOpenVDB.cmake" in CMAKE_MODULE_PATH this project has
23:25:39   asked CMake to find a package configuration file provided by "OpenVDB", but
23:25:39   CMake did not find one.
23:25:39 
23:25:39   Could not find a package configuration file provided by "OpenVDB" with any
23:25:39   of the following names:
23:25:39 
23:25:39     OpenVDBConfig.cmake
23:25:39     openvdb-config.cmake
23:25:39 
23:25:39   Add the installation prefix of "OpenVDB" to CMAKE_PREFIX_PATH or set
23:25:39   "OpenVDB_DIR" to a directory containing one of the above files.  If
23:25:39   "OpenVDB" provides a separate development package or SDK, be sure it has
23:25:39   been installed.
23:25:39 Call Stack (most recent call first):
23:25:39   /opt/ros/jazzy/share/openvdb_vendor/cmake/openvdb_vendorConfig.cmake:41 (include)
23:25:39   CMakeLists.txt:31 (find_package)
23:25:39 
23:25:39 
23:25:39 -- Configuring incomplete, errors occurred!
```

Although the Humble and Iron are working.  I am seeing a [successful ARM openvdb_vendor build](https://build.ros2.org/job/Jbin_unv8_uNv8__openvdb_vendor__ubuntu_noble_arm64__binary/) along with Humble and Iron, so it seems that the issue is unique to Jazzy & x86 works fine. I see that #303 fixed the ARM builds of openvdb_vendor. The error its reporting is [from this line](https://github.com/SteveMacenski/spatio_temporal_voxel_layer/blob/jazzy/openvdb_vendor/openvdb_vendor-extras.cmake#L8) as STVL is finding the openvdb_vendor package. #301 avoids compiling OpenVDB on Jazzy or newer when it is available from apt in those distributions. Is OpenVDB not on ARM on 24.04? `<depend condition="$ROS_DISTRO != humble and $ROS_DISTRO != iron">libopenvdb-dev</depend>` should be getting it. Can you check that you have `libopenvdb-dev` on your computer (apt install / apt show)?

I'm curious if you build STVL locally using the openvdb_vendor binaries if that works. If not, that's a good sign since that means we can debug this problem locally and thus more easily. 

I think perhaps the problem could be that we missed adding `list(APPEND CMAKE_MODULE_PATH "/usr/lib/aarch64-linux-gnu/cmake/OpenVDB")` in `openvdb_vendor/openvdb_vendor-extras.cmake`. If you look at #301 and #303, 301 adds 2x of those `list` calls, but 303 only adds in the ARM call for 1x of them.

agoeckner, 2024-11-05:
Hey Steve, no worries. Would've liked to see you at IROS but I think you chose the less oppressive option :)

----

I will check whether the openvdb-dev exists later today.

Interestingly though, our CI fails to build STVL with this error, indicating that it can't find OpenVDB:

````
#12 [4/5] RUN cd "/opt/robot" && . /opt/ros/jazzy/setup.sh && colcon build 
#12 39.37 Starting >>> apriltag_msgs
#12 39.90 Starting >>> grex_network_interfaces
#12 40.18 Starting >>> explore_lite
#12 40.49 Starting >>> ld08_driver
#12 40.79 Starting >>> openvdb_vendor
#12 41.08 Starting >>> turtlebot3_description
#12 41.37 Starting >>> apriltag
#12 41.65 Starting >>> laser_filters
#12 88.05 [Processing: apriltag, apriltag_msgs, explore_lite, grex_network_interfaces, laser_filters, ld08_driver, openvdb_vendor, turtlebot3_description]
#12 88.05 Finished <<< turtlebot3_description [47.0s]
#12 92.22 --- stderr: openvdb_vendor
#12 92.22 COMPONENT = openvdb
#12 92.22 ---
#12 92.22 Finished <<< openvdb_vendor [51.4s]
#12 92.24 Starting >>> spatio_temporal_voxel_layer
#12 148.0 [Processing: apriltag, apriltag_msgs, explore_lite, grex_network_interfaces, laser_filters, ld08_driver, spatio_temporal_voxel_layer]
#12 148.0 Finished <<< apriltag [1min 47s]
#12 158.2 --- stderr: ld08_driver
#12 158.2 /opt/robot/src/ld08_driver/src/transform.cpp: In constructor ‘SlTransform::SlTransform(LDVersion, bool)’:
#12 158.2 /opt/robot/src/ld08_driver/src/transform.cpp:30:50: warning: unused parameter ‘to_right_hand’ [-Wunused-parameter]
#12 158.2    30 | SlTransform::SlTransform(LDVersion version, bool to_right_hand)
#12 158.2       |                                             ~~~~~^~~~~~~~~~~~~
#12 158.2 ---
#12 158.2 Finished <<< ld08_driver [1min 58s]
#12 158.3 Starting >>> turtlebot3_bringup
#12 201.8 [Processing: apriltag_msgs, explore_lite, grex_network_interfaces, laser_filters, spatio_temporal_voxel_layer, turtlebot3_bringup]
#12 201.8 Finished <<< turtlebot3_bringup [43.6s]
#12 280.3 [Processing: apriltag_msgs, explore_lite, grex_network_interfaces, laser_filters, spatio_temporal_voxel_layer]
#12 280.3 [Processing: apriltag_msgs, explore_lite, grex_network_interfaces, laser_filters, spatio_temporal_voxel_layer]
#12 280.3 Finished <<< explore_lite [4min 0s]
#12 349.4 [Processing: apriltag_msgs, grex_network_interfaces, laser_filters, spatio_temporal_voxel_layer]
#12 349.4 [Processing: apriltag_msgs, grex_network_interfaces, laser_filters, spatio_temporal_voxel_layer]
#12 349.4 Finished <<< laser_filters [5min 8s]
#12 356.5 --- stderr: spatio_temporal_voxel_layer
#12 356.5 CMake Warning (dev) at /usr/lib/aarch64-linux-gnu/cmake/pcl/Modules/FindFLANN.cmake:45 (find_package):
#12 356.5   Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
#12 356.5   variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
#12 356.5   cmake_policy command to set the policy and suppress this warning.
#12 356.5 
#12 356.5   CMake variable FLANN_ROOT is set to:
#12 356.5 
#12 356.5     /usr
#12 356.5 
#12 356.5   For compatibility, find_package is ignoring the variable, but code in a
#12 356.5   .cmake module might still use it.
#12 356.5 Call Stack (most recent call first):
#12 356.5   /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:261 (find_package)
#12 356.5   /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:306 (find_flann)
#12 356.5   /usr/lib/aarch64-linux-gnu/cmake/pcl/PCLConfig.cmake:570 (find_external_library)
#12 356.5   /opt/ros/jazzy/share/pcl_conversions/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
#12 356.5   /opt/ros/jazzy/share/pcl_conversions/cmake/pcl_conversionsConfig.cmake:41 (include)
#12 356.5   CMakeLists.txt:22 (find_package)
#12 356.5 This warning is for project developers.  Use -Wno-dev to suppress it.
#12 356.5 
#12 356.5 CMake Error at /opt/robot/install/openvdb_vendor/share/openvdb_vendor/cmake/openvdb_vendor-extras.cmake:8 (find_package):
#12 356.5   By not providing "FindOpenVDB.cmake" in CMAKE_MODULE_PATH this project has
#12 356.5   asked CMake to find a package configuration file provided by "OpenVDB", but
#12 356.5   CMake did not find one.
#12 356.5 
#12 356.5   Could not find a package configuration file provided by "OpenVDB" with any
#12 356.5   of the following names:
#12 356.5 
#12 356.5     OpenVDBConfig.cmake
#12 356.5     openvdb-config.cmake
#12 356.5 
#12 356.5   Add the installation prefix of "OpenVDB" to CMAKE_PREFIX_PATH or set
#12 356.5   "OpenVDB_DIR" to a directory containing one of the above files.  If
#12 356.5   "OpenVDB" provides a separate development package or SDK, be sure it has
#12 356.5   been installed.
#12 356.5 Call Stack (most recent call first):
#12 356.5   /opt/robot/install/openvdb_vendor/share/openvdb_vendor/cmake/openvdb_vendorConfig.cmake:41 (include)
#12 356.5   CMakeLists.txt:31 (find_package)
#12 356.5 
#12 356.5 
#12 356.5 ---
#12 356.5 Failed   <<< spatio_temporal_voxel_layer [4min 24s, exited with code 1]
#12 466.4 Aborted  <<< apriltag_msgs [7min 7s]
#12 591.2 Aborted  <<< grex_network_interfaces [9min 11s]
#12 592.2 
#12 592.2 Summary: 7 packages finished [9min 26s]
#12 592.2   1 package failed: spatio_temporal_voxel_layer
#12 592.2   2 packages aborted: apriltag_msgs grex_network_interfaces
#12 592.2   3 packages had stderr output: ld08_driver openvdb_vendor spatio_temporal_voxel_layer
#12 592.2   5 packages not processed
#12 ERROR: process "/bin/sh -c cd \"$ROBOT_WS\" && . /opt/ros/jazzy/setup.sh && colcon build $ROBOT_COLCON_ARGS" did not complete successfully: exit code: 1
````

SteveMacenski, 2024-11-05:
I assume you run rosdep before building? STVL / openvdb vendor should be grabbing that or building it if it needs to. Though 

```
#12 92.22 --- stderr: openvdb_vendor
#12 92.22 COMPONENT = openvdb
#12 92.22 ---
```

might be diagonstic

agoeckner, 2024-11-05:
The libopenvdb-dev package does indeed exist for arm64: https://packages.ubuntu.com/noble/libopenvdb-dev

I can confirm that it's installed in my Ubuntu 24.04 Docker image (probably from rosdep).

agoeckner, 2024-11-05:
Yeah, that resolved it. I submitted a PR in #306.

agoeckner, 2024-11-05:
Awesome, thanks for the help!

SteveMacenski, 2024-11-06:
Sync happening now!

closed 2024-11-06 by SteveMacenski

agoeckner, 2024-11-18:
For some reason, the STVL package still doesn't exist in the binary-arm64 APT repository: http://packages.ros.org/ros2/ubuntu/dists/noble/main/binary-arm64/Packages

I'm not sure what the pipeline looks like for this repository or else I would investigate it myself.

SteveMacenski, 2024-11-18:
https://build.ros2.org/search/?q=spatio_temporal&Jenkins-Crumb=c8d9d5599b9005a6133a88444a89d44cb2beaec6e08366cf5ab95596173a12a2

They're all turning over. The distribution releases need to happen so that the staging binaries are put into the main apt index for general users. You can use it today using `testing` apt rather than `main`. See https://docs.ros.org/en/rolling/Installation/Testing.html

tl;dr: its out of our hands now, we're just waiting for distribution syncs to occur, which do about every 4-6 weeks. You can track them here: https://discourse.ros.org/c/release/16

agoeckner, 2024-11-18:
Thanks for the explanation!

PR #306 (2024-11-05): Add arm64 version of OpenVDB to CMake module path.
This resolves #305.

As stated in https://github.com/SteveMacenski/spatio_temporal_voxel_layer/issues/305#issuecomment-2455928201, the CMake module path was not updated completely in #303 for arm64 builds.